raspibolt/raspibolt #6: raspberry bitcoin docker

lucaszanella, 2018-03-18:
Wouldn't it be better to dockerize the bitcoin lightning tutorial? 

It'd be just a matter of mounting the external drive correctly and setting up passwords and private keys. I see that you require to add networking information before the first boot. I'm doing an SD file tree editor for raspberry pi that automates this entire part and installs things (like docker) at first run: https://github.com/lucaszanella/raspberry_script_setup

I'm just waiting for a pruned mode to be launched, then I'll certainly make a tutorial based on this one, but it'd be much easier if dockerized

Stadicus, 2018-03-18:
Hi Lucas, yes a docker image probably would be most efficient way for spinning up new nodes. This is not my goal however, as it involves trusting someone else and does not provide the same learning experience as doing it myself.

Thanks for the link to the Raspi Script Setup. Interesting, will check it out! 

If you would like to dockerize my setup: feel free. Pleaae keep me posted.

lucaszanella, 2018-03-19:
I didn't understand the part about trusting someone else. You mean trusting the docker base images are secure, or trusting by doing docker run without reviewing?

In the second case, I usually package everything in my Dockerfile, no dependencies on other files, so it looks like a regular bash script that's easy to review. I even add SHA verifications when I use wget inside the dockerfile. 

In the first case, I think the most used images like ubuntu can be audited now.

But I agree that lazy users would simply run the dockerfile without any reviewing.  

Anyways, I'm going to dockerize everything once pruning mode becomes available :)

robclark56, 2018-03-26:
Personally - I enjoyed this as a learning experience. Glad Stadicus explained every step, rather than just saying 'download and run this script'.

Both approaches have their place.

Stadicus, 2018-03-26:
Agree. This is more the "learning and getting my hands dirty" approach, while the docker approach is obviously much more efficient. Different goals, different guides.

closed 2018-03-26 by Stadicus