raspibolt/raspibolt #8: admin.macaroon missing after installing lnd

hardcqre, 2018-03-18:
Hi first of all thanks for the great guide!
i followed it to the letter but am stuck at creating a new address for the testnet coins (after creating the wallet with "lncli create")
lncli misses the admin.macaroon which should be in /home/bitcoin/.lnd/admin.macaroon but it just isnt there.

btw: I copied all the scripts

Did i miss a step? I just can't figure it out.

Stadicus, 2018-03-18:
Not sure what could cause this, no special steps are necessary. LND create these files automatically, as long as

- lnd is up and running (as "admin", check `sudo journalctl -u lnd`)
- you are using `lncli` with the user "bitcoin" (user "admin" must not work)
- file permissions are set correctly (check screenshot): everything should be owned by "bitcoin"

[image]

hardcqre, 2018-03-19:
Thanks for your reply.
However my settings were exactly like that.
I managed to fix it though.

After compiling lnd as described in the lnd repo didn't work either. i deleted the wallet.db in .lnd/data/..../wallet.db and created a new wallet.
This time i set a passphrase for the seed which i thought was optional and didn't do the first time.
It now created the files and syncs just fine. weird.

You could maybe add a common issue section. : )

cootpursuits, 2018-03-19:
I have the same issue. Deleting the wallet.db and creating a new one with a seed passphrase did not solve it.

hardcqre, 2018-03-19:
did you stop lnd and restart it?

cootpursuits, 2018-03-19:
Yes, it didn't help, neither did restarting the whole raspi.

Are you sure it wasn't compiling lnd as per the lnd repo after all that made it work for you?

hardcqre, 2018-03-19:
it could play a part in the solution although it didn't work until i
created a new wallet and deleting the old one beforehand.

Stadicus, 2018-03-19:
The binaries should create the `admin.macaroon`, no build is required. IMHO, it has no connection to the wallet.db at all. Definitely not depending on a seed password. Strangely, I never had any issues with it before.

I was able to found an old issue that my give some additional hints. 
https://github.com/lightningnetwork/lnd/issues/279

Additionally, I opened an issue with LND to verify my installation description. Maybe I missed something? 
https://github.com/lightningnetwork/lnd/issues/890

hardcqre, 2018-03-20:
Thanks stadicus. I can only speak from my experience. After i unlocked The
New wallet with lncli unlock, The macaroons got written in .lnd/. Not
before. I especially checked that, hence why i mentioned it AS my solution.
After reinstalling lnd via the github repo and starting lnd, The macaroons
were still missing.

What caused it, i dont know.

lightningnetwork/lnd#279

lightningnetwork/lnd#890

cryptopacker, 2018-03-20:
Great guide -- very thorough and easy to follow. 

I have the same problem as the OP. When running <code>lncli newaddress np2wkh</code>, I get **[lncli] open /home/bitcoin/admin.macaroon: no such file or directory.**  

File permissions are identical to the screenshot above. However, there is no **readonly.macaroon**. The output of <code>sudo journalctl -u</code> shows that LN is up and running. 

Should I try deleting the macaroons.db  in   /.lnd/data and restarting?

Then when I try to unlock the wallet with "lncli unlock" and enter the LN password, I get **[lncli] rpc error: code = Unknown desc = missing address manager namespace**

If I were to try hardcqre's suggestion, how would I delete the existing wallet? Sorry if that's a silly question, as normally I'd just delete the wallet.dat file, but I can't find it in the .lnd/data folder. Yet I know the wallet exists because when I go to create a wallet, it says it already exists!

Two other quick questions regarding the install:

1. When setting up the LND config file, it says to "adjust to your alias". Could someone elaborate on that, please? What should I change that to?

2. Towards the end of syncing the bitcoin blockchain I got a warning: "Warning: unknown new rules activated (versionbit 28)" I've done a cursory search and it seems not to be a cause for concern? Would you agree?

I'm still a bit of a noob in this space, but I am eager. Any help would be much appreciated, as I am very excited to get my LN toes wet and lend a helping hand in spreading the LN word.

cootpursuits, 2018-03-20:
As for deleting macaroons.db and restarting, I tried that, and the macaroons.db gets created again, but still no admin.macaroon or readonly.macaroon.

I've read in another issue that they might be created in a different directory, but I did a `sudo find / -name "*.macaroon"`, and there's nothing there.

GeorgeSpark-Stahl, 2018-03-20:
I'm not sure I can provide much more insight into this, other than I'm having an issue similar to @cootpursuits where I deleted the lnd data directory and now only macaroons.db is getting recreating.  I've also searched around to see if the files were created elsewhere.  This is on macOS High Sierra.

Stadicus, 2018-03-20:
I don't think I can be of any help here. May I recommend to share your experience in the official LND issue I opened? 

https://github.com/lightningnetwork/lnd/issues/890

Stadicus, 2018-03-20:
Ok, here are the *exact* steps I took to set up a brand new installation of LND on testnet, binary downloaded as described in the guide. I could not reproduce the error for a brand new installation. Everything working fine.

As mentioned in the LND issue, when moving to mainnet, the files `admin.macaroon`, `readonly.macaroon` and `data/macaroons.db` should be deleted. I will update my guide in this regard today.

Hope that helps.

```
admin@RaspiBolt:~ $ ls -la /home/bitcoin/.lnd/
total 12
drwxr-xr-x 2 bitcoin bitcoin 4096 Mar 20 21:42 .
drwxr-xr-x 6 bitcoin bitcoin 4096 Mar 20 21:39 ..
-rw-r--r-- 1 bitcoin bitcoin  390 Mar 20 21:42 lnd.conf
admin@RaspiBolt:~ $ cat /home/bitcoin/.lnd/lnd.conf
# RaspiBolt LND Mainnet: lnd configuration
# /home/bitcoin/.lnd/lnd.conf

[Application Options]
debuglevel=debug
debughtlc=true
maxpendingchannels=5
alias=YOUR_NAME [LND]
color=#68F442

[Bitcoin]
bitcoin.active=1

# enable either testnet or mainnet
bitcoin.testnet=1
#bitcoin.mainnet=1

bitcoin.node=bitcoind

[autopilot]
autopilot.active=1
autopilot.maxchannels=5
autopilot.allocation=0.6

admin@RaspiBolt:~ $ cat /etc/systemd/system/lnd.service
# RaspiBolt LND Mainnet: systemd unit for lnd
# /etc/systemd/system/lnd.service

[Unit]
Description=LND Lightning Daemon
Requires=bitcoind.service
After=getpublicip.service
After=bitcoind.service

# for use with sendmail alert
#OnFailure=systemd-sendmail@%n

[Service]
# get var PUBIP from file
EnvironmentFile=/run/publicip

ExecStart=/usr/local/bin/lnd --externalip=${PUBLICIP}
PIDFile=/home/bitcoin/.lnd/lnd.pid
User=bitcoin
Group=bitcoin
Type=simple
KillMode=process
TimeoutSec=180
Restart=always
RestartSec=60

[Install]
WantedBy=multi-user.target

admin@RaspiBolt:~ $ sudo systemctl start lnd

admin@RaspiBolt:~ $ systemctl status lnd
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-03-20 21:46:24 CET; 6s ago
 Main PID: 7450 (lnd)
   CGroup: /system.slice/lnd.service
           └─7450 /usr/local/bin/lnd --externalip=91.190.22.151

admin@RaspiBolt:~ $ ls -la /home/bitcoin/.lnd/
total 28
drwxr-xr-x 4 bitcoin bitcoin 4096 Mar 20 21:46 .
drwxr-xr-x 6 bitcoin bitcoin 4096 Mar 20 21:39 ..
drwx------ 3 bitcoin bitcoin 4096 Mar 20 21:46 data
-rw-r--r-- 1 bitcoin bitcoin  390 Mar 20 21:42 lnd.conf
drwx------ 3 bitcoin bitcoin 4096 Mar 20 21:46 logs
-rw-r--r-- 1 bitcoin bitcoin  709 Mar 20 21:46 tls.cert
-rw------- 1 bitcoin bitcoin  227 Mar 20 21:46 tls.key

admin@RaspiBolt:~ $ sudo journalctl -f -u lnd
-- Logs begin at Tue 2018-03-20 20:57:29 CET. --
Mar 20 21:46:25 RaspiBolt lnd[7450]: Attempting automatic RPC configuration to bitcoind
Mar 20 21:46:25 RaspiBolt lnd[7450]: Automatically obtained bitcoind's RPC credentials
Mar 20 21:46:25 RaspiBolt lnd[7450]: 2018-03-20 21:46:25.195 [INF] LTND: Version 0.4.0-alpha
Mar 20 21:46:25 RaspiBolt lnd[7450]: 2018-03-20 21:46:25.196 [INF] LTND: Active chain: Bitcoin (network=testnet)
Mar 20 21:46:25 RaspiBolt lnd[7450]: 2018-03-20 21:46:25.210 [INF] CHDB: Checking for schema update: latest_version=0, db_version=0
Mar 20 21:46:25 RaspiBolt lnd[7450]: 2018-03-20 21:46:25.210 [INF] RPCS: Generating TLS certificates...
Mar 20 21:46:25 RaspiBolt lnd[7450]: 2018-03-20 21:46:25.220 [INF] RPCS: Done generating TLS certificates
Mar 20 21:46:25 RaspiBolt lnd[7450]: 2018-03-20 21:46:25.250 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
Mar 20 21:46:25 RaspiBolt lnd[7450]: 2018-03-20 21:46:25.252 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Mar 20 21:46:25 RaspiBolt lnd[7450]: 2018-03-20 21:46:25.252 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create wallet, or `lncli unlock` to unlock already created wallet.
^C

admin@RaspiBolt:~ $ sudo su bitcoin

bitcoin@RaspiBolt:/home/admin $ cd

bitcoin@RaspiBolt:~ $ lncli create
Input wallet password:
Confirm wallet password:

Do you have an existing cipher seed mnemonic you want to use? (Enter y/n): n

Your cipher seed can optionally be encrypted.
Input your passphrase you wish to encrypt it (or press enter to proceed without a cipher seed passphrase):

Generating fresh cipher seed...

!!!YOU MUST WRITE DOWN THIS SEED TO BE ABLE TO RESTORE THE WALLET!!!

---------------BEGIN LND CIPHER SEED---------------
...
...
...
---------------END LND CIPHER SEED-----------------

!!!YOU MUST WRITE DOWN THIS SEED TO BE ABLE TO RESTORE THE WALLET!!!

lnd successfully initialized!

bitcoin@RaspiBolt:~ $ ls -la /home/bitcoin/.lnd/
total 36
drwxr-xr-x 4 bitcoin bitcoin 4096 Mar 20 21:49 .
drwxr-xr-x 6 bitcoin bitcoin 4096 Mar 20 21:39 ..
-rw------- 1 bitcoin bitcoin  207 Mar 20 21:49 admin.macaroon
drwx------ 4 bitcoin bitcoin 4096 Mar 20 21:48 data
-rw-r--r-- 1 bitcoin bitcoin  390 Mar 20 21:42 lnd.conf
drwx------ 3 bitcoin bitcoin 4096 Mar 20 21:46 logs
-rw-r--r-- 1 bitcoin bitcoin  164 Mar 20 21:49 readonly.macaroon
-rw-r--r-- 1 bitcoin bitcoin  709 Mar 20 21:46 tls.cert
-rw------- 1 bitcoin bitcoin  227 Mar 20 21:46 tls.key

bitcoin@RaspiBolt:~ $
```

aakselrod, 2018-03-20:
If someone who's having this issue can show logs similar to the above for reproducing it, that would be extremely helpful.

cryptopacker, 2018-03-21:
I managed to solve my LN wallet creation issues by deleting the two folders in the data directory and essentially starting the wallet creation process again, (following hardcqre's suggestion above and also recommended here: https://github.com/lightningnetwork/lnd/issues/408). I then rebooted (not sure if that was necessary), made sure LND was running, ran <code>lncli create </code> and followed the instructions in the guide. I did add a passphrase this time. Everything working now. Thanks.

cootpursuits, 2018-03-21:
Yesterday at one point I somehow got the macaroons created, then I tried to reproduce the issue, but I wasn't able to ever not get the macaroons created again. Today I started from scratch, and sure enough the issue resurfaced.

@aakselrod here are my logs, looks to me like lnd is killed before creating the macaroons:

```
admin@RaspiBolt:~ $ ls -la /home/bitcoin/.lnd/
total 12
drwxr-xr-x 2 bitcoin bitcoin 4096 Mar 21 10:08 .
drwxr-xr-x 5 bitcoin bitcoin 4096 Mar 21 09:42 ..
-rw-r--r-- 1 bitcoin bitcoin  387 Mar 21 09:42 lnd.conf

admin@RaspiBolt:~ $ cat /home/bitcoin/.lnd/lnd.conf
# RaspiBolt LND Mainnet: lnd configuration
# /home/bitcoin/.lnd/lnd.conf

[Application Options]
debuglevel=debug
debughtlc=true
maxpendingchannels=5
alias=RaspiBolt
color=#68F442

[Bitcoin]
bitcoin.active=1

# enable either testnet or mainnet
bitcoin.testnet=1
#bitcoin.mainnet=1

bitcoin.node=bitcoind

[autopilot]
autopilot.active=1
autopilot.maxchannels=5
autopilot.allocation=0.6

admin@RaspiBolt:~ $ cat /etc/systemd/system/lnd.service
# RaspiBolt LND Mainnet: systemd unit for lnd
# /etc/systemd/system/lnd.service

[Unit]
Description=LND Lightning Daemon
Requires=bitcoind.service
After=getpublicip.service
After=bitcoind.service

# for use with sendmail alert
#OnFailure=systemd-sendmail@%n

[Service]
# get var PUBIP from file
EnvironmentFile=/run/publicip

ExecStart=/usr/local/bin/lnd --externalip=${PUBLICIP}
PIDFile=/home/bitcoin/.lnd/lnd.pid
User=bitcoin
Group=bitcoin
Type=simple
KillMode=process
TimeoutSec=180
Restart=always
RestartSec=60

[Install]
WantedBy=multi-user.target

admin@RaspiBolt:~ $ sudo systemctl start lnd

admin@RaspiBolt:~ $ systemctl status lnd
● lnd.service - LND Lightning Daemon
[... 65 lines cut ...]
bitcoin@RaspiBolt:~ $ systemctl status lnd
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2018-03-21 10:11:08 CET; 16s ago
  Process: 1760 ExecStart=/usr/local/bin/lnd --externalip=${PUBLICIP} (code=exited, status=2)
 Main PID: 1760 (code=exited, status=2)
 
bitcoin@RaspiBolt:~ $ exit
exit

admin@RaspiBolt:~ $ sudo journalctl -f -u lnd
-- Logs begin at Thu 2016-11-03 18:16:42 CET. --
Mar 21 10:12:08 RaspiBolt systemd[1]: Stopped LND Lightning Daemon.
Mar 21 10:12:08 RaspiBolt systemd[1]: Started LND Lightning Daemon.
Mar 21 10:12:08 RaspiBolt lnd[1821]: Attempting automatic RPC configuration to bitcoind
Mar 21 10:12:08 RaspiBolt lnd[1821]: Automatically obtained bitcoind's RPC credentials
Mar 21 10:12:08 RaspiBolt lnd[1821]: 2018-03-21 10:12:08.679 [INF] LTND: Version 0.4.0-alpha
Mar 21 10:12:08 RaspiBolt lnd[1821]: 2018-03-21 10:12:08.680 [INF] LTND: Active chain: Bitcoin (network=testnet)
Mar 21 10:12:08 RaspiBolt lnd[1821]: 2018-03-21 10:12:08.680 [INF] CHDB: Checking for schema update: latest_version=0, db_version=0
Mar 21 10:12:08 RaspiBolt lnd[1821]: 2018-03-21 10:12:08.730 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
Mar 21 10:12:08 RaspiBolt lnd[1821]: 2018-03-21 10:12:08.732 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Mar 21 10:12:08 RaspiBolt lnd[1821]: 2018-03-21 10:12:08.732 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create wallet, or `lncli unlock` to unlock already created wallet.
^C
admin@RaspiBolt:~ $
```

hardcqre, 2018-03-21:
@cootpursuits try unlocking The created wallet. This created my macaroons. It May be that you have to create and directly after creation unlock it. At least it was The case in my side

cootpursuits, 2018-03-21:
This is what I get, no macaroons created:

```
bitcoin@RaspiBolt:~/.lnd $ lncli unlock
Input wallet password:
[lncli] rpc error: code = Unknown desc = missing address manager namespace
```

cootpursuits, 2018-03-21:
What made it work in the end was deleting everything in `~/.lnd` except `lnd.conf`, restarting both bitcoind and lnd, and creating a new wallet. Doing the same without restarting bitcoind did not work.

aakselrod, 2018-03-21:
@cootpursuits it looks like `lnd` crashes because it runs out of RAM trying to generate an encryption key from your password. That can take over 250MB for a few seconds.

Stadicus, 2018-03-21:
Not sure why this keeps happening just for a few people, but constantly, but not for most, as all are using the same setup and hardware. Is everyone using a Raspberry Pi 3 here? 

What is the free Memory just before creating the wallet? (check with `htop`)

For me, the CPU load during regular operations (here on testnet) with both bitcoind and lnd is minimal. Memory usage is also very reasonable (539 MB free):

[image]

ZebulonFrantzich, 2018-03-25:
@Stadicus This issue has been resolved for me. I reflashed Raspbian Lite and walked through the guide multiple times with the same result of missing .macaroon files. Once the guide was changed to move the Swap File to the external hard disk, the .macaroon files were generated on that iteration through the guide. Thank you for looking into this!

Stadicus, 2018-03-25:
That supports the "not enough RAM" thesis, with the swap file acting as an additional buffer.

closed 2018-03-25 by Stadicus

ZebulonFrantzich, 2018-03-27:
@Stadicus I agree with the "not enough RAM" thesis as well. I think the problem was that when I ran 'Expand Filesystem' in 'raspi-config', I received the following error: <code>Re-reading the partition table failed.: Device or resource busy</code> I am guessing that did not allow enough memory, so the additional buffer was required. Thank you for this guide!

nipponese, 2018-06-18:
I had the same issue. There is another raspi tutorial floating around that suggest removing the swapfile all together. After re-install dphys-swap LND was able to generate the macaroon files.

OlaStenberg, 2018-06-22:
Thanks @Stadicus, spend hours trying to fix this. Can also confirm that @Stadicus solution worked by clearing swap files. Got it working on my stationary PC which is runnin WSL. I am assuming it will fix my raspberry as well, will edit with an update tomorrow.

rek79, 2018-09-24:
Resolved the issue (I think).  It does not seem to be RAM related at all.  If you are configured for testnet then, **by default**, LND will look in `~/.lnd/data/chain/bitcoin/**testnet**/admin.macaroon` and `~/.lnd/data/chain/bitcoin/**testnet**/readonly.macaroon` See [https://github.com/lightningnetwork/lnd/blob/master/sample-lnd.conf].  I've confirmed that the macaroon files are indeed created in the default testnet directories after running `lncli create`.  However, when executing `lncli newaddress np2wkh` the error reads `[lncli] unable to read macaroon path (check the network setting!): open /home/bitcoin/.lnd/data/chain/bitcoin/**mainnet**/admin.macaroon: no such file or directory`.  So, it is looking in mainnet for the macaroon files.  I'm not sure how to fix this other than override the macaroon default paths in lnd.conf.  Any suggestions on keeping defaults but having it work on testnet?

bruceleeroy18, 2019-01-02:
Any change in this issue? I am currently having the same problem when attempting to run the lncli getinfo command on Litecoin Mainnet. I have put macaroonpath=~/data/chain/litecoin/mainnet/admin.macaroon but the error message says lncli is still looking for a bitcoin/mainnet directory.

phptek, 2019-01-08:
I had the same problem and it was due to stale tls certs in LND's home dir. Removing these and restaring lnd, regenerated the macaroons in the data dir. Hope this helps someone else.

bruceleeroy18, 2019-01-08:
Thanks! 
After a reboot, if I use --chain=litecoin for all commands it recognizes the lncli commands. 
I will attempt to regenerate new tls certs in the home dir later today. 

Sent via the Samsung Galaxy S7 edge, an AT&T 4G LTE smartphone
-------- 原始訊息 --------自： Russ Michell <notifications@github.com> 日期： 2019/1/8  12:55  (GMT-05:00) 至： Stadicus/guides <guides@noreply.github.com> 副本： bruceleeroy18 <browningsmith@gmail.com>, Comment <comment@noreply.github.com> 主旨： Re: [Stadicus/guides] admin.macaroon missing after installing lnd (#8) 
I had the same problem and it was due to stale tls certs in LND's home dir. Removing these and restaring lnd, regenerated the macaroons in the data dir. Hope this helps someone else.

—
[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Stadicus/guides/issues/8#issuecomment-452391756",
"url": "https://github.com/Stadicus/guides/issues/8#issuecomment-452391756",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]