raspibolt/raspibolt #11: Raspbolt

Snead, 2018-03-21:
In the guide for raspberry you make this steps:
sudo systemctl enable bitcoind
sudo shutdown -r now

But you don't take into account that we have synced datadir and it kills the chainstate so you have to reindex.

Stadicus, 2018-03-21:
You're referring to the [Bitcoin] section, right? The command `sudo systemctl enable bitcoind.service` does not yet start bitcoind, in fact it is started with the reboot for the first time. 

Later on, systemd handles the bitcoind shutdown gracefully when restarting, the timeout should be generous enough. 

I don't see where a chainstate corruption could occur. Am I missing something?

Snead, 2018-03-22:
yeah that is actually weird but i had no started bitcoin and when i enabled it it actually started.
im on raspberry pi 3 with raspbian lite.

Stadicus, 2018-03-22:
Can you reproduce this?

closed 2018-03-25 by Stadicus