raspibolt/raspibolt #13: autopilot not working

cootpursuits, 2018-03-22:
Should the autopilot be working with the binaries used in the guide? It doesn't for me, I'm having this issue: https://github.com/lightninglabs/lightning-app/issues/26

Stadicus, 2018-03-22:
Yes, it's working for me. Once everything is fully synced, LND starts opening channels one by one. I never compiled the beta, only using the binary.

Stadicus, 2018-03-22:
Maybe connect to a few nodes to get network info? 

You can always disable autopilot in `lnd.conf`and create your channels manually.

cootpursuits, 2018-03-22:
Interesting. I'm synced to the chain, the wallet has a confirmed balance, I manually connected to the nodes suggested in the other thread, and I'm seeing the peers, yet there are no channels opened.

closed 2018-03-31 by Stadicus