raspibolt/raspibolt #15: Lightning Wallet

meeark, 2018-03-23:
Hi,  when initially trying to unlock my new lightning wallet i'm getting a message saying wallet is encrypted. I'm assuming i'm putting in the wrong password, I must have accidentally mispelled it twice when creating it. How do I go about setting up a new one?
Thanks,

Stadicus, 2018-03-23:
I once ran into that issue as well. First, try to just restart lnd with `sudo systemctl restart lnd` and unlock the wallet again.

If that does not help, you can just stop lnd, delete everything in the `/home/bitcoin/.lnd` folder except the `lnd.conf`, restart lnd, let it sync and create a new wallet.

Maybe there are more precise ways to just delete the wallet, feel free to try, but this is the most thorough way to reset lnd (especially, as the macaroon files and macaroons.db are also encrypted with your wallet password).

closed 2018-03-31 by Stadicus

UnholyLizard, 2018-04-01:
Same as 

> when initially trying to unlock my new lightning wallet i'm getting a message saying wallet is encrypted. 

Restart lnd with `sudo systemctl restart lnd` helped me