raspibolt/raspibolt #17: No password input when entering lnd unlock password, remotely off site, in Putty

JohnWarneGates, 2018-03-23:
I'm not sure if this is an issue with Putty settings or LND.

I'm not home this weekend, so I setup ufw to accept the IP where I'm staying. I signed in fine on admin. I got into user bitcoin fine, with the admin password, yet, when i do lncli unlock, the password prompt comes up but it seems no input is being accepted to the Putty shell.

After entering the password, correctly, I get the prompt:

"[lncli] Wallet is encrypted. Please unlock using 'lncli unlock', or set password using 'lncli create' if this is the first time starting lnd."

As if I didn't enter any input, whatsoever. Putty doesn't act this way, locally, when I'm at home. It goes through, no problem.

Does anyone out there know what could be the problem? I didn't want to post this to the LND issues page because I'm a total newb when it comes to Linux/Unbuntu and this could be a common problem with beginners and, imo, it would be better served here,

Dropin16, 2018-03-25:
Restart Lightning with `sudo systemctl restart lnd`

Then try logging in again with  `lncli unlock`

Now you should get the message : lnd successfully unlocked!

JohnWarneGates, 2018-03-25:
Thanks, Dropin16! Worked like a charm!

closed 2018-03-25 by JohnWarneGates

Stadicus, 2018-03-25:
@Dropin16 , could you elaborate on what the exact issue is here?

Stadicus, 2018-03-25:
Thanks for your feedback, I will make the guide a bit clearer in that regard.