raspibolt/raspibolt #18: After unlock - No connection available

sobchek, 2018-03-25:
Very helpful walkthrough. Thanks.

Created passwords and seeds, all smooth. Upon attempting to create a new address, or even getinfo, I am getting this error.

> [lncli] rpc error: code = Unavailable desc = grpc: the connection is unavailable

I checked the lnd log:

> 2018-03-25 00:00:33.154 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
> 2018-03-25 00:00:33.154 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create wallet, or `lncli unlock` to unlock already created wallet.
> 2018-03-25 00:07:13.458 [INF] LNWL: Opened wallet
> 2018-03-25 00:07:13.921 [ERR] SRVR: invalid password
> 2018-03-25 00:08:14.567 [INF] LTND: Version 0.4.0-alpha
> 2018-03-25 00:08:14.567 [INF] LTND: Active chain: Bitcoin (network=testnet)
> 2018-03-25 00:08:14.591 [INF] CHDB: Checking for schema update: latest_version=0, db_version=0
> 2018-03-25 00:08:14.618 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
> 2018-03-25 00:08:14.619 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
> 2018-03-25 00:08:14.619 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create wallet, or `lncli unlock` to unlock already created wallet.

I can go on an unlocking, successfully, and then attempting to create a new address, but this cycle keeps happening.

Help would be much appreciated.

gg2001, 2018-03-25:
I am getting the same thing.

Stadicus, 2018-03-25:
The log says `invalid password`, any chance of a mixup or typo? 

If nothing helps (and you do not have any funds commited yet), I would recommend deleting the `/mnt/hdd/lnd` directory, recreating it with the config file, let sync `lnd` directly on mainnet and creating a brand-new wallet. 

The reason for a complete delete is that quite a few files are encrypted using this password (eg. *.macaroon, data/macaroons.db and of course the wallet.db in the chains subdirectories). You can try to delete these individually, but I would just start anew.

sobchek, 2018-03-25:
Wiped the slate clean and started from the beginning, (had done this prior to original post also.)

I can unlock the wallet. No problem. lnd.log:

> Mar 25 16:05:51 raspberrypi lnd[793]: 2018-03-25 16:05:51.929 [INF] LNWL: Opened wallet
> Mar 25 16:05:52 raspberrypi lnd[793]: 2018-03-25 16:05:52.335 [INF] LTND: Primary chain is set to: bitcoin
> Mar 25 16:05:52 raspberrypi lnd[793]: 2018-03-25 16:05:52.335 [INF] LTND: Initializing bitcoind backed fee estimator
> Mar 25 16:05:58 raspberrypi lnd[793]: 2018-03-25 16:05:58.598 [INF] LNWL: Opened wallet
> Mar 25 16:05:58 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
> Mar 25 16:05:58 raspberrypi systemd[1]: lnd.service: Unit entered failed state.
> Mar 25 16:05:58 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.

sooo... checked my lnd.service file. The initial issue I see is the reference to the lnd.pid file, but there is no corresponding file in that directory.

Why or how? No idea, hopefully someone can help.

Thanks

sobchek, 2018-03-25:
Ok... fresh install. Same problem. Came to realize that I cannot, at least with this setup, simultaneously run bitcoind mainnet and testnet, which is where I believe the problem was. I was just assuming testnet3 was running as planned, but it was giving me an error about bitcoin.conf, and I didn't bother to listen.

So changed bitcoin.conf back to testnet / reboot / boots into testnet / lncli unlock / now catching up...

I assume this has resolved the issue, though still catching up.

closed 2018-03-26 by sobchek