raspibolt/raspibolt #19: Can't use bitcoin-cli unless restart

robclark56, 2018-03-25:
shutdown -r now
Login as admin

admin ~ ฿ sudo su bitcoin

bitcoin /home/admin ฿ systemctl status bitcoind
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset:
   Active: active (running) since Sun 2018-03-25 17:39:23 AWST; 12min ago
  Process: 539 ExecStart=/usr/local/bin/bitcoind -pid=/home/bitcoin/.bitcoin/bit
 Main PID: 555 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─555 /usr/local/bin/bitcoind -pid=/home/bitcoin/.bitcoin/bitcoind.pi

bitcoin /home/admin ฿ tail -f /home/bitcoin/.bitcoin/testnet3/debug.log
[...]
2018-03-25 09:55:58 UpdateTip: new best=00000000794c2c0158a5e35820aec55cc84bdceb4a992154042b509a76756e2c height=1289135 version=0x20000000 log2_work=69.810808 tx=17649803 date='2018-03-25 09:55:51' progress=1.000000 cache=0.1MiB(543txo)

[Bitcoind is Running - OK  (testnet), and synced]

bitcoin /home/admin ฿ bitcoin-cli getblockchaininfo
error: couldn't connect to server: unknown (code -1)
(make sure server is running and you are connecting to the correct RPC port)

bitcoin /home/admin ฿ systemctl restart bitcoind

bitcoin /home/admin ฿  bitcoin-cli getblockchaininfo
{
  "chain": "test",
  "blocks": 1289135,
[...]
  },
  "warnings": ""
}

Now all good.

100% reproducable
????

robclark56, 2018-03-25:
PS ... Absolutely GREAT guide by the way!

Thanks

Stadicus, 2018-03-25:
What's your output of `sudo netstat -nap|grep bitcoin|grep LISTEN`?

johnny-cash, 2018-03-25:
In service getpublicip

Change 

After=network.target

Stadicus, 2018-03-25:
I changed this setting as there is a known issue with the "After=network.target" that would result in exactly this. If you `run sudo netstat -nap|grep bitcoin|grep LISTEN`, you should see tcp bound to both ports :8332 and :8333 (either on ip 127.0.0.1 or 0.0.0.0).

If the suggestion above does not work, try the setting `After=multi-user.target` in `getpublicip.service`.

johnny-cash, 2018-03-25:
The issue with "After=network.target" is solved change in raspi-config the boot options to wait for network at boot.

"After=network.target" + "Wait for network at boot" in raspi-config -> work 100%

Stadicus, 2018-03-25:
I for myself never had this issue, I am not sure when this happens as I cannot reproduce it. @robclark56, your feedback on the above options would be very helpful.

Stadicus, 2018-03-25:
> "After=network.target" + "Wait for network at boot" in raspi-config

I think, the combination provided by @johnny-cash is the most robust one. I included it in the guide.

robclark56, 2018-03-26:
Good news. Thanks team.

Making these 2 changes fixed the problem

- Updating getpublicip.service with "After=network.target"

- "Wait for network at boot" in raspi-config

BONUS
It also solved a related issue with lnd not starting with an externalip
BEFORE
systemctl status lnd
/usr/local/bin/lnd --externalip=<blank>

AFTER
systemctl status lnd
/usr/local/bin/lnd --externalip=AAA.BBB.CCC.DDD  <--- my external IP

:-)

closed 2018-03-26 by robclark56