raspibolt/raspibolt #22: Sending payment not working

Dropin16, 2018-03-25:
My node is online:
[https://explorer.acinq.co/#/n/0366e4191c0271562700edfcfadca560c1a999b0b76731cf97eed1c5695bc418d9](url)

Got 3 peers and 6 Channels.

Balance is as following:     
"total_balance": "61109819",
"confirmed_balance": "61109819",
"unconfirmed_balance": "0"

When making a transaction I input following:
``$ lncli sendpayment --pay_req=lntb19u1pdt07h5pp5ra6qa7hmpd3qatr.........07g46mu97u5s2q2jsyjvg`

The issue is that nothing happens after this step. The "cursor" just stays one line below without the prefix 
`admin@AURORA:~ $`

I can press Ctrl+C and then enter `$ lncli listpayments`

Result is:     "payments": [
    ]

Tried 5 times with different payment requests.

What may the reason be that I can't perform payments?

raindogdance, 2018-03-25:
Does this happen with all payments you are trying to send? I had a similar issue with some certain nodes, the Blockstream node in particular (in the LND Slack was suggested that it is because they are running an older version of c-lightning, which causes the issue). Perhaps try to get/send a tip here: http://tip.bight.nl - the maintainer has a LND node.  

Also, perhaps try `lncli payinvoice <paymentrequest>` instead?

Dropin16, 2018-03-25:
It happens with Y'alls and Starblocks. 

Tried  http://tip.bight.nl without adding node infos or anything besides the amount of satoshis and got this message: 
` lncli sendpayment --pay_req=lnbc1u1pdtsqlcpp5sg7xe76q...............qp7jv8u9`
[lncli] rpc error: code = Unknown desc = unknown network

Same Error when adding my own node's ID in the node description.

Tried `lncli payinvoice <paymentrequest>` and got the following:

{
        "payment_error": "unable to route payment to destination: UnknownNextPeer",
        "payment_preimage": "",
        "payment_route": null
}

raindogdance, 2018-03-25:
Oh, are you on testnet or mainnet? Starblocks is definitely testnet and Yalls has both versions (depending on their URL). "Unknown network" might suggest you are on testnet trying to pay a mainnet invoice?

Dropin16, 2018-03-25:
I'll drop a part of the log in here in case it helps. Can always delete it if it doesn't help.

Mar 25 22:11:20 AURORA lnd[20662]: 2018-03-25 22:11:20.756 [INF] PEER: unable to read message from 34.201.11.250:46062: read tcp 192.168.1.130:9735->34.201.11.250:46062: use of closed network connection
Mar 25 22:11:20 AURORA lnd[20662]: 2018-03-25 22:11:20.756 [INF] HSWC: Removing channel link with ChannelID(eb16e993ccf7cc679c01198a2b5def9f322100b293abbc07cae566225046a4c6)
Mar 25 22:11:20 AURORA lnd[20662]: 2018-03-25 22:11:20.759 [INF] HSWC: ChannelLink(c6a446502266e5ca07bcab93b20021329fef5d2b8a19019c67ccf7cc93e916eb:0) is stopping
Mar 25 22:11:20 AURORA lnd[20662]: 2018-03-25 22:11:20.761 [INF] NTFN: Cancelling epoch notification, epoch_id=195
Mar 25 22:11:20 AURORA lnd[20662]: 2018-03-25 22:11:20.762 [INF] HSWC: ChannelLink(c6a446502266e5ca07bcab93b20021329fef5d2b8a19019c67ccf7cc93e916eb:0) has exited
Mar 25 22:11:21 AURORA lnd[20662]: 2018-03-25 22:11:21.126 [ERR] HSWC: unable to find links by destination unable to locate channel link bydestination hop id 022bc10a5df9d410edf883916e08e2ed35f5ea948d3e3e16c3e15fba6cbdd5287d
Mar 25 22:11:21 AURORA lnd[20662]: 2018-03-25 22:11:21.127 [ERR] CRTR: Attempt to send payment 1f75291ee123b8db41d46d1ab52195a90e7b0962b03bebae289d6d7de7599228 failed: UnknownNextPeer
Mar 25 22:11:22 AURORA lnd[20662]: 2018-03-25 22:11:22.209 [INF] HSWC: Sent 0 satoshis received -304 satoshis in the last 10 seconds (1.8446744073709553e+18 tx/sec)
Mar 25 22:11:22 AURORA lnd[20662]: 2018-03-25 22:11:22.447 [ERR] CRTR: Attempt to send payment 1f75291ee123b8db41d46d1ab52195a90e7b0962b03bebae289d6d7de7599228 failed: TemporaryChannelFailure
Mar 25 22:11:22 AURORA lnd[20662]: 2018-03-25 22:11:22.580 [INF] DISC: Broadcasting batch of 3 new announcements
Mar 25 22:11:23 AURORA lnd[20662]: 2018-03-25 22:11:23.806 [ERR] CRTR: Attempt to send payment 1f75291ee123b8db41d46d1ab52195a90e7b0962b03bebae289d6d7de7599228 failed: UnknownNextPeer
Mar 25 22:11:25 AURORA lnd[20662]: 2018-03-25 22:11:25.789 [ERR] CRTR: Attempt to send payment 1f75291ee123b8db41d46d1ab52195a90e7b0962b03bebae289d6d7de7599228 failed: UnknownNextPeer
Mar 25 22:11:27 AURORA lnd[20662]: 2018-03-25 22:11:27.825 [ERR] CRTR: Attempt to send payment 1f75291ee123b8db41d46d1ab52195a90e7b0962b03bebae289d6d7de7599228 failed: TemporaryChannelFailure
Mar 25 22:11:29 AURORA lnd[20662]: 2018-03-25 22:11:29.907 [ERR] CRTR: Attempt to send payment 1f75291ee123b8db41d46d1ab52195a90e7b0962b03bebae289d6d7de7599228 failed: UnknownNextPeer
Mar 25 22:11:31 AURORA lnd[20662]: 2018-03-25 22:11:31.976 [ERR] CRTR: Attempt to send payment 1f75291ee123b8db41d46d1ab52195a90e7b0962b03bebae289d6d7de7599228 failed: UnknownNextPeer
Mar 25 22:11:32 AURORA lnd[20662]: 2018-03-25 22:11:32.209 [INF] HSWC: Sent 0 satoshis received 0 satoshis in the last 10 seconds (1.2 tx/sec)
Mar 25 22:11:33 AURORA lnd[20662]: 2018-03-25 22:11:33.260 [ERR] CRTR: Attempt to send payment 1f75291ee123b8db41d46d1ab52195a90e7b0962b03bebae289d6d7de7599228 failed: UnknownNextPeer
Mar 25 22:11:34 AURORA lnd[20662]: 2018-03-25 22:11:34.571 [ERR] CRTR: Attempt to send payment 1f75291ee123b8db41d46d1ab52195a90e7b0962b03bebae289d6d7de7599228 failed: TemporaryChannelFailure
Mar 25 22:11:35 AURORA lnd[20662]: 2018-03-25 22:11:35.795 [ERR] CRTR: Attempt to send payment 1f75291ee123b8db41d46d1ab52195a90e7b0962b03bebae289d6d7de7599228 failed: TemporaryChannelFailure

Dropin16, 2018-03-25:
Ooooh yes I'm on mainnet. I'll try a Y'alls payment with your method.

raindogdance, 2018-03-25:
Are you sure you are on mainnet? You linked to your node on the Acinq explorer in the OP, and that explorer is definitely for testnet.

Dropin16, 2018-03-25:
I'm so sorry my brain had a fatal error when typing. I'm on testnet! 

I get:

    "identity_pubkey": "0366e4191c0271562700edfcfadca560c1a999b0b76731cf97eed1c5695bc418d9",
    "alias": "AURORA",
    "num_pending_channels": 2,
    "num_active_channels": 3,
    "num_peers": 6,
    "block_height": 1289172,
    "block_hash": "00000000de8de71b9e0d97b56a641d53ef59d4eab20dab2b74604d54731dde87",
    "synced_to_chain": true,
    "testnet": true,
    "chains": [
        "bitcoin"
    ],
    "uris": [
        "0366e4191c0271562700edfcfadca560c1a999b0b76731cf97eed1c5695bc418d9@85.1.111.22:9735"
    ],
    "best_header_timestamp": "1522009807"

Dropin16, 2018-03-25:
Y'alls payment with `lncli payinvoice <paymentrequest>` and got no reaction like in the first post.

raindogdance, 2018-03-26:
I'm afraid my knowledge is limited here. Hopefully someone with more understanding will chime in. The only thing that comes to mind to try is a good old client restart (`sudo systemctl start lnd`) and/or to open up a second ssh window, log in to your Pi and start a realtime log file: `sudo journalctl -f -u lnd`, while trying the sending command in the first window, and try to make sense of what is going on.  
There is a lot of chatter in the log file though, so probably this won't make much sense... So I am hoping someone else will have a better idea.

Stadicus, 2018-03-26:
As you are up and running and LND is synced to the chain, this is somewhat outside the scope this guide. I would suggest you 

* ask in the LND slack channels ([invite link](http://dev.lightning.community/resources/) on the lnd resources page)
* open an issue in the LND github repo, but make sure to document it thoroughly

closed 2018-03-26 by Dropin16

Stadicus, 2018-03-26:
Great to hear. I wouldn't give too much thought regarding the Explorer visualization. It's not very accurate and also contains stale information. It's nice, nontheless. :-)