raspibolt/raspibolt #23: Channel OFFLINE if LND Wallet Locked

robclark56, 2018-03-26:
Not sure if this an issue with LND, or something that can be fixed in RaspiBolt

ISSUE
The Eclair Testnet Wallet App sees a channel that exists to a RaspiBolt as OFFLINE if the LND wallet has not been unlocked. This is an issue as that means if a RaspiBolt just boots up (or recovers from a power failure) then the channel remains offline.

NETWORK

[LIGHTNING TESTNET] ---฿฿฿฿--- [My RaspiPi] ---฿฿฿฿--- [Eclair]

---฿฿฿฿--- = Funded channel

DETAILS
* Fresh bootup of RaspiBolt 
* bitcoind and lnd running fine (testnet)
* lnd wallet not unlocked
* Android Phone with the Eclair Wallet (https://play.google.com/store/apps/details?id=fr.acinq.eclair.wallet&hl=en)
* Eclair has one open (and funded) channel with my RaspiBolt. 
* Eclair channel Status = OFFLINE
* Can't pay  for Coffee at https://starblocks.acinq.co/#/ from Eclair
* lncli unlock 
* <wait 60 sec>
* Eclair channel Status = NORMAL
* Successfully pay for Coffee at https://starblocks.acinq.co/#/ from Eclair, (routed through my RaspiBolt) 

[image]

Stadicus, 2018-03-26:
AFAIK this is exactly how it's supposed to work. For now, I set up an email alert that sends a message every time the service is started (not yet in the guide, a bit complicated with Gmail certs & stuff).

I am not sure yet, if this can be automated in a safe way.

robclark56, 2018-03-27:
Thanks for the answers. So this prompted me to find the 'default' PW code in LND ... which is below.

// We wait until the user provides a password over RPC. In case lnd is
// started with the --noencryptwallet flag, we use the default password
// "xxxx" for wallet encryption.
privateWalletPw := []byte("xxxxx")
publicWalletPw := []byte("xxxxx")

[Note: I replaced the actual PWs with xxxxx for this post .... which is a bit paranoid I will admit as the LND  code is public anyway. But my point is that I now know what the default PWs are ... and anybody that can use Google can learn them too].

What this says is that all std LND implementations use the same default PWs if --noencryptwallet is used (e.g. to unlock the wallet automatically at bootup). Seems like a security hole big enough to drive a bus through, like routers left at factory default with login/pw = admin/admin.

Hmmm. How to proceed?
1) I don't want to use --noencryptwallet as the world now knows my wallet PW
2) But want to have my LN node run 'lights off', including after a power failure.

Guess this is why this is called BETA testing

:-)

Appreciate this discussion belongs elsewhere ... but adding this post for completeness.

Stadicus, 2018-03-27:
Thanks for sharing this additional data. In my understanding, there is no way to securely store the "secret" for unlocking the wallet on the same machine. If the system is breached, the attacker would gain access to both and could easily run off with the funds.

Therefore, either unlock manually, or let a second system with a secure key storage unlock the node with RPC remotely. This makes sense in my opinion, but I'm not sure myself how I want to proceed.

closed 2018-06-13 by Stadicus