raspibolt/raspibolt #24: Does RaspiBolt support Dynamic public IP?

robclark56, 2018-03-28:
I understand that getpublicip.service  updates /run/publicip with the public IP every 10 mins.

What I am unclear about is ... if the Public IP changes after startup, does lnd get restarted with that new publicip?

- Yes? : Great. Which code is doing the restart?
- No?:  Not much point in updating  /run/publicip every 10 mins

The reason I ask is that I do not have a static IP address from my ISP and I use a service like www.dyn.com to give me a static [FQDN](https://en.wikipedia.org/wiki/Fully_qualified_domain_name) instead.

If RaspiBolt does not support dynamic external IP addresses - then I was thinking of developing a script to notice if  /run/publicip changed ... and then restart lnd.  But, that brings up the ugly issue of the wallet needs to be unlocked again. :-(

Stadicus, 2018-03-28:
You are bringing up valid points, I am not sure I cab address them all.

Afaik, the public IP is only needed to announce your node, not to transact with open channels. But not quite sure. 

Also, there iy no easy way to update your public IP without restarting the service atm, which leads to the (in this context unnecessary) issue of thr locked wallet. So, no, the ip is not changed after the service is started.

A great way for you to discuss these topics is the LND slack. See link at the bottom of http://dev.lightning.community

Would appreciate if you keep us posted here with any findings

Stadicus, 2018-03-31:
Would be great if you could formalize your work in a pull request.

robclark56, 2018-03-31:
Update:

I now have my 'Lights Off' RaspiBolt working as I want

- Power cycle boots up with an unlocked wallet

- If my ISP changes my IP address, lnd restarts wit the new --externalip

Want to test more deeply before documenting in a Pull request

sobchek, 2018-04-03:
Curious... I can’t seem to bootstrap from DNS seeds, as intended. Could this be related to my dynamic IP?

sobchek, 2018-04-04:
Great thanks. I had tried manually connecting previously with no success. I will update tonight and give it another go.

Stadicus, 2018-06-13:
Support for NAT traversal and dynamic IP changes has just been merged into LND master trunk and will probably be available with the next release.
https://github.com/lightningnetwork/lnd/pull/1109

closed 2018-06-13 by Stadicus