raspibolt/raspibolt #26: Hardening Pi

Ldngzero, 2018-03-29:
Hi, I cannot install firewall or fail2ban.
admin@raspi:~ $ sudo apt-get install ufw
[sudo] password for admin:
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package ufw is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or is only available from another source
E: Package 'ufw' has no installation candidate

admin@raspi:~ $ sudo apt-get install fail2ban
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package fail2ban

Thanks for your great work

Stadicus, 2018-03-29:
What Linux distro are you using? Did you update your package lists with `sudo apt update`?

Ldngzero, 2018-03-29:
I did sudo apt update again and now all worked!
I am using Debian 2018-03-13-raspbian-stretch-lite.

Thanks again

closed 2018-03-29 by Ldngzero