raspibolt/raspibolt #34: Could not login as lncli listchannels would hang

robclark56, 2018-04-01:
Reporting here in case it happens to someone else.

For reasons I have yet to understand, right now on my RaspiBolt, this command is hanging
_lncli listchannels_

As a consequence, that means that the raspibolt script (admin MOTD) was hanging => I could not get a prompt when logging in as admin. 

I have seen reports from others on Google of _lncli listchannels_ hanging.

I managed to get to the prompt (by rebooting and logging in ASAP) and commented out this line (for now) so at least I can login

Commented out this line in /usr/local/bin/raspibolt
#ln_channels_total="........."

Still investigating but so far ......

A) 
lncli connect xxxx@xxx.xxx.xxx.xxx:9735
[lncli] rpc error: code = Unknown desc = chain backend is still syncing, server not active yet

B)
bitcoin /home/admin ฿ bitcoin-cli getblockchaininfo
{
  "verificationprogress": 0.9999970407145348  <--- this number is **DECREASING** over time
}

C) 
bitcoin /home/admin ฿ lncli getinfo
{
    "num_pending_channels": 0,
    "num_active_channels": 0,
    "num_peers": 0,                         <----------- ????
    "synced_to_chain": false,            <----------  ????
    "testnet": true,
    "chains": [
        "bitcoin"
    ],
}

Stadicus, 2018-04-01:
Added `.hushlogin` for root as fallback login possibility: https://github.com/Stadicus/guides/commit/0c957f6c26072c6a0410dd4645e539521433ce4f

robclark56, 2018-04-02:
Could not solve.

Reinstalling from scratch.

closed 2018-04-02 by robclark56

robclark56, 2018-05-09:
FWIW ... again these 2 commands are hanging for me

lncli listchannels
lncli getinfo

But many others are fine
lncli walletbalance
lncli channelbalance
lncli listinvoices
etc

* lnd bug?
* Issue with 32-bit Raspian OS?

Reporting for info only.

tomav, 2019-04-12:
I have exactly the same issue.
@robclark56 have you found the source of the issue?

robclark56, 2019-04-12:
Interestingly - today my raspibolt was 'hanging' on lncli commands

$sudo systemctl stop lnd
$sudo systemctl restart bitcoind
$sudo sytemctl start lnd
$lncli unlock

... fixed it for me.