raspibolt/raspibolt #37: Insecure permissions when moving Swap File

Argonath, 2018-04-01:
Not able to enable new swap configuration because of insecure permissions.

When executing the command  `sudo dphys-swapfile swapon`, the following error occurs:

```
swapon: /mnt/hdd/swapfile: insecure permissions 0777, 0600 suggested.
swapon: /mnt/hdd/swapfile: read swap header failed
```

My node's `dphys-swapfile` file (using default settings only changing `CONF_SWAPFILE ` and `CONF_SWAPSIZE`:

```
  GNU nano 2.7.4                   File: /etc/dphys-swapfile

# /etc/dphys-swapfile - user settings for dphys-swapfile package
# author Neil Franklin, last modification 2010.05.05
# copyright ETH Zuerich Physics Departement
#   use under either modified/non-advertising BSD or GPL license

# this file is sourced with . so full normal sh syntax applies

# the default settings are added as commented out CONF_*=* lines

# where we want the swapfile to be, this is the default
#CONF_SWAPFILE=/var/swap
CONF_SWAPFILE=/mnt/hdd/swapfile

# set size to absolute value, leaving empty (default) then uses computed value
#   you most likely don't want this, unless you have an special disk situation
CONF_SWAPSIZE=1000

# set size to computed value, this times RAM size, dynamically adapts,
#   guarantees that there is enough swap without wasting disk space on excess
#CONF_SWAPFACTOR=2

# restrict size (computed and absolute!) to maximally this limit
#   can be set to empty for no limit, but beware of filled partitions!
#   this is/was a (outdated?) 32bit kernel limit (in MBytes), do not overrun it
#   but is also sensible on 64bit to prevent filling /var or even / partition
#CONF_MAXSWAP=2048
```

Stadicus, 2018-04-01:
Try setting the permissions `0700` for the swapfile with the following command (if the file has already been created): `sudo chmod 0700 /mnt/hdd/swapfile`

Some additional infos:
http://raspberrypimaker.com/adding-swap-to-the-raspberrypi/

Argonath, 2018-04-04:
Changing the permision to `0600` using `sudo chmod 0600 /mnt/hdd/swapfile` didn't actually solve the problem, I'm still getting:

```
swapon: /mnt/hdd/swapfile: insecure permissions 0777, 0600 suggested.
swapon: /mnt/hdd/swapfile: read swap header failed
```
I don't know why the permission is not changing, I have to check what may be the cause. Anyway, my node still up and running, but not sure what will happen if the swapfile persists in this unknown situation.

Perhaps it is just a problem with my node, not sure if other people are experiencing it, I wouldn't change the guide yet.

Stadicus, 2018-04-17:
Will close for now, probably the same issue as in #53, but please reopen if you cannot solve it.

closed 2018-04-17 by Stadicus