raspibolt/raspibolt #38: EPS download releases

chris-belcher, 2018-04-01:
Thanks for creating a guide for Electrum Personal Server. I have an edit but not sure of the exact form it should take.

Right now the guide instructs users to clone the github repository, but this gives them the latest master branch. That might contain bugfixes, works-in-progress and untested-features. Ideally users would instead download a tagged release from https://github.com/chris-belcher/electrum-personal-server/releases

Problem is, each release has a different file name. The guide could instruct users to `wget https://github.com/chris-belcher/electrum-personal-server/archive/v0.1.1.zip` and then `unzip v0.1.1.zip` but then the directory inside will be called `electrum-personal-server-v0.1.1`. Those version numbers will change at every release. All this may actually cause a headache for people following that guide.

So I'm not sure how to write that part of the guide. Ideally users would download releases, but also the guide should hopefully be as simple as possible to follow. So what's the best way to do this?

Stadicus, 2018-04-04:
Hi Chris, thank you for the greate EPS and your quick response last weekend. I will update the guide in the near future, but that can take some time. Or you can create a pull request. :-)

I think the way to go forward is to let the directory contain the version number, but to create a generig sym link (eg. /home/bitcoin/eps/) to this directory. Works like a charm also for bitcoind and lnd (both stored on the external hdd).

Stadicus, 2018-05-24:
I updated the guide to check out a specific tag from your releases page. Is this a good way forward in your oppinion?
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_64_electrum.md

chris-belcher, 2018-05-25:
Yes that should work

Stadicus, 2018-05-26:
Great, thanks!

closed 2018-05-26 by Stadicus