raspibolt/raspibolt #39: Fully trustless bitcoin nodes

chris-belcher, 2018-04-01:
A couple of times this guide uses the phrase "fully trustless bitcoin node". Especially in the [bitcoin node page](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_30_bitcoin.md) it says that using a full node makes bitcoin trustless. This is true but incomplete, and it might be worth rephrasing.

Full nodes check all the validity rules of bitcoin. Miners set the history of transactions.

Bitcoin's security model requires both full nodes and miners to be decentralized. The full-node-using economy must be decentralized to stop fake bitcoins being accepted as payment, the miners must be decentralized to stop censorship of transactions and to make transactions irreversible.

In your guide you could say something like "full nodes are required to make bitcoin trustless". Maybe mention this analogy I like: "If bitcoin is digital gold, then a full node wallet is your own personal goldsmith who checks for you that received payments are genuine" or "If bitcoin is digital cash, then a full node wallet is your own personal laboratory which checks for you that received banknotes are real."

Stadicus, 2018-04-04:
Valid points, I fully agree. I'm not that much of a writer, your input is very welcome. Would you like to propose these changes? Otherwise, I'll update the wording when I come around to it.

Stadicus, 2018-05-24:
Incorporated both your comments into the intro page, could not have phrased it better. Thanks!

closed 2018-05-24 by Stadicus