raspibolt/raspibolt #49: LN cant start 'runtime: out of memory'

UnholyLizard, 2018-04-04:
bitcoind fully synced (testnet)

Memory usage:
```
              total        used        free      shared  buff/cache   available
Mem:           976M        101M        818M        1.0M         56M        826M
Swap:          2.0G        380M        1.6G

```

After LN unlock:
```
Apr 04 23:42:41 RaspiB lnd[6528]: 2018-04-04 23:42:41.766 [INF] LNWL: The wallet has been unlocked without a time limit
Apr 04 23:42:41 RaspiB lnd[6528]: 2018-04-04 23:42:41.766 [INF] LTND: LightningWallet opened
Apr 04 23:44:20 RaspiB lnd[6528]: runtime: out of memory: cannot allocate 374341632-byte block (1032159232 in use)
Apr 04 23:44:20 RaspiB lnd[6528]: fatal error: out of memory
Apr 04 23:44:20 RaspiB lnd[6528]: runtime stack:
Apr 04 23:44:20 RaspiB lnd[6528]: runtime.throw(0x8c7f9c, 0xd)
Apr 04 23:44:20 RaspiB lnd[6528]:         /usr/local/go/src/runtime/panic.go:619 +0x60
Apr 04 23:44:20 RaspiB lnd[6528]: runtime.largeAlloc(0x16500000, 0x60101, 0x76f39000)
Apr 04 23:44:20 RaspiB lnd[6528]:         /usr/local/go/src/runtime/malloc.go:828 +0xdc
Apr 04 23:44:20 RaspiB lnd[6528]: runtime.mallocgc.func1()
Apr 04 23:44:20 RaspiB lnd[6528]:         /usr/local/go/src/runtime/malloc.go:721 +0x38
Apr 04 23:44:20 RaspiB lnd[6528]: runtime.systemstack(0x0)
Apr 04 23:44:20 RaspiB lnd[6528]:         /usr/local/go/src/runtime/asm_arm.s:349 +0x80
Apr 04 23:44:20 RaspiB lnd[6528]: runtime.mstart()
Apr 04 23:44:20 RaspiB lnd[6528]:         /usr/local/go/src/runtime/proc.go:1170
Apr 04 23:44:20 RaspiB lnd[6528]: goroutine 84 [running]:
Apr 04 23:44:20 RaspiB lnd[6528]: runtime.systemstack_switch()
Apr 04 23:44:20 RaspiB lnd[6528]:         /usr/local/go/src/runtime/asm_arm.s:294 +0x4 fp=0x12f4ac54 sp=0x12f4ac50 pc=0x68c0c
Apr 04 23:44:20 RaspiB lnd[6528]: runtime.mallocgc(0x16500000, 0x81f080, 0x4282a601, 0x179f4)
Apr 04 23:44:20 RaspiB lnd[6528]:         /usr/local/go/src/runtime/malloc.go:720 +0x86c fp=0x12f4acb4 sp=0x12f4ac54 pc=0x21130
Apr 04 23:44:21 RaspiB systemd[1]: lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Apr 04 23:44:21 RaspiB systemd[1]: lnd.service: Unit entered failed state.
Apr 04 23:44:21 RaspiB systemd[1]: lnd.service: Failed with result 'exit-code'.
```
Memory usage at time around error:
```
              total        used        free      shared  buff/cache   available
Mem:           976M        916M         39M        672K         20M         25M
Swap:          2.0G        596M        1.4G
```
Reboot, restart, not helped.

Stadicus, 2018-04-05:
What version are you running, already on 0.4.1-beta? Do you start `lnd` directly from the command line or using `systemd`? 

This looks like a program internal error to me. I would open a issue with `lnd` directly so you can provide the relevant logs to the devs.

UnholyLizard, 2018-04-06:
At start it was 0.4.0 and all worked fine.
Possible after not correctly shutdown (weak/bad power supply on RaspPi or external HDD or both) LND can't start after wallet unlock.
Upgrade to 0.4.1 beta didn't help.

`lnd` started by `systemd`

Yes, thank you. I just open issue with lnd github.

closed 2018-04-06 by UnholyLizard