raspibolt/raspibolt #53: sudo rm /var/swap ERROR

jackwatson86, 2018-04-06:
Hi guys,

I should start by saying I'm not especially tech savvy and this is the first time I've tried doing something like this. So far the guide has been super straightforward and I'm enjoying the process, but now I've hit a wall.

When I run `sudo rm /var/swap` I get the following in return:

rm: cannot remove '/var/swap':  Operation not permitted

Anyone know how to get past this?

Thanks

Argonath, 2018-04-07:
Maybe related to #37. I'm also having some trouble regarding permission when creating the swap.

UnholyLizard, 2018-04-08:
Well I played with `dphys-swapfile` a bit and noticed if just change `sudo nano /etc/dphys-swapfile` with new param `CONF_SWAPFILE` and do `sudo dphys-swapfile setup` this new swap may be "added" to the old one that you can't delete. May be reboot will help, but my weird (I think something is wrong there) solution was:
`$ sudo nano /etc/dphys-swapfile`
Chage `CONF_SWAPFILE=/var/swap` save, exit
`$ sudo dphys-swapfile uninstall` 
`$ sudo nano /etc/dphys-swapfile`
Chage `CONF_SWAPFILE=/mnt/hdd/swapfile` save, exit
`$ sudo dphys-swapfile setup`
`$ sudo dphys-swapfile swapon`
Now try `$ sudo rm /var/swap` If I remember correctly that file was already deleted in my case.

Stadicus, 2018-04-12:
I think that the `sudo dphys-swapfile swapon` command was missing before deleting the old swap file. 

Commit https://github.com/Stadicus/guides/commit/47fa9da57d78cfa3344316d90c9d1df8ea075429

closed 2018-04-17 by Stadicus

jackwatson86, 2018-04-17:
Thanks @Stadicus this seems to have worked! Onwards with the guide! :)

elkimek, 2018-07-01:
Hey, I'm having similar issue but just with another permission.

when I do:

```
sudo dphys-swapfile swapon
```
I get:

```
swapon: /mnt/hdd/swapfile: insecure permissions 0775, 0600 suggested.
swapon: /mnt/hdd/swapfile: read swap header failed
```

I tried all abovem mentioned incl.

```
sudo chmod 0600 /mnt/hdd/swapfile
```
and didn't help.

I didn't  have this problem when I was following
[Option 1 (recommended): Format hard disk with Ext4](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#option-1-recommended-format-hard-disk-with-ext4)

But problem appeared first when I went with opt. 2:
[Option 2: Use existing hard disk with NTFS](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#option-2-use-existing-hard-disk-with-ntfs)