raspibolt/raspibolt #56: Verification of bitcoind operations (testnet)

Cees2daBees, 2018-04-11:
Great guide!! I'm a total noob and have been following along and feeling like I'm making pretty good progress.

I have run into an issue though under the section Bitcoin -> Verification of bitcoind operations. Note: running on testnet.

When I run the command..

$ sudo tail -f /home/bitcoin/.bitcoin/testnet3/debug.log

..it returned multiple lines that read "Potential stale tip detected, will try using extra outbound peer" and

$ bitcoin-cli getblockchaininfo

showed blocks are 0 and headers are 0 and verificationprogress was not really moving.

A quick google search lead me to a potential solution where I added the following lines to the bitcoin.conf file:

addnode=104.237.131.138
addnode=151.80.205.132
addnode=192.155.82.123
addnode=74.220.255.190
addnode=80.100.203.151

And now it seems it might be syncing.

My question is, is this the proper way of solving this, and will this fix hold up when I switch over to mainnet?

Thanks!

Stadicus, 2018-04-12:
Seems you had problems establishing your very first connections with other bitcoin nodes to get information about the network and other peers. In that case it's the right thing to do to just add one or more random peers manually, bitcoind will take it from there.

Could you post the link to article / post where you found this solution?

Cees2daBees, 2018-04-12:
Ok, thanks. I'll take a look again when I've switched over to mainnet.

Link to the post where I found this solution is below:

https://bitcoin.stackexchange.com/questions/73607/testnet3-node-sync-process-is-very-slow?utm_medium=organic&utm_source=google_rich_qa&utm_campaign=google_rich_qa

Oh, and side note.. just bought my Blockaccino last night! Super fast and 1 sat fee! =)

Thanks for this guide, it is awesome! Can't wait to sync the Bitcoin blockchain so I could jump over to mainnet!

closed 2018-04-12 by Cees2daBees