raspibolt/raspibolt #57: Updates to new LND versions

raindogdance, 2018-04-14:
More a question than an "issue". 
I want to update from LND 0.4.0 to the newest version. Do I understand it right that for that I only need to run the install steps as described here: https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_40_lnd.md#install-lnd ? Probably would need to stop the LND service first, just to be sure?

Perhaps it would be good to add a note to the guide for people who are starting off on older versions (and also the advice to check for newer version every once in a while).  

Also, after the update, should I be deleting the files in the Pi's "download" folder, so the HDD space doesn't get cluttered over the long run?

Stadicus, 2018-04-14:
Yes, thats a good point, I will include a comment regarding updating the version (or you might want to create a pull request to speed things up? :-).

You can just stop LND with `sudo systemctl stop lnd`, but you should then delete the macaroon files in your LND directory with `sudo rm /home/bitcoin/.lnd/*.macaroon`. This needs to be done as the upgrade provides a new macaroon (see related issue https://github.com/lightningnetwork/lnd/issues/921), otherwise you can no longer create invoices.

Then just reinstall everything according to the guide, no need to delete old stuff. You're free to do so, of course, but it's not that big.

After that, just start LND again with `sudo systemctl start lnd`, unlock your wallet, and copy the `admin.macaroon` to the admin LND directory again. 

I'm typing this just out of memory, let me know if something does not work or if additional steps are needed.

raindogdance, 2018-04-15:
Thanks!

I had to start the `wget` and `tar` commands with `sudo`, because otherwise I got a "permission denied" messages. Seeing that that chapter of the guide starts off with "logging as admin user" (and I am logged in as "Pi"), I'm guessing it's because I set up my Pi with an older version of the guide, so probably the `sudo`won't be necessary for users who followed the newer version?

Otherwise the update seems to worked. I haven't yet copied the `admin.macaroon` file anywhere yet, because I don't know how (what command to use and which directory?), but the client hasn't been complaining yet, so I'm assuming this is not critical/necessary?

Stadicus, 2018-04-15:
Yes, I got rid of the `pi` user to make the guide more easily extendable with other hardware like the Odroid HC2. The `admin` user is the the new `pi` user. :-)

With that in mind, you can copy the macaroon file with the following command (maybe replace the "admin" directory with "pi")

`sudo cp /home/bitcoin/.lnd/admin.macaroon /home/admin/.lnd/`

raindogdance, 2018-04-15:
Hm, I don't seem to have the /home/pi/.lnd/ directory (nor a /home/admin/.lnd/ one). When I do `cd /home` and `ls -la`, I see the `pi` folder there, but when I go `cd /home/pi/` it just takes me to the "root" directory again and `cd /home/pi/.lnd/` says `no such file or directory` (and the same if I put `admin` instead of `pi`)

Stadicus, 2018-04-17:
The `/home/admin/.lnd` directory comes from the guide here: 
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_40_lnd.md#assign-lnd-permissions-to-admin

This way, you don't have to start a `bitcoin` user session everytime and `bitcoin-cli` and `lncli` can be used by the `admin` user.

Stadicus, 2018-05-24:
No, channels will remain active as long as you just install the new update and keep everything else intact.

waffi121, 2018-05-31:
I have just updated. LND stopped, new LND installed, LND re-enabled. Everything works! Just crazy. : D thanks Stadicus:)

Stadicus, 2018-06-24:
Update procedure is now part of the FAQ:
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_faq.md#how-to-upgrade-lnd-bin-

closed 2018-06-24 by Stadicus