raspibolt/raspibolt #61: error: couldn't connect to server: unknown (code -1)

nperalez, 2018-04-19:
Hello,

First off, thank you so much for writing this. I'm not entirely new to running a full node as this is now my third time but I'm having the following problem:

When I run

$ systemctl status bitcoind.service

My screenshot seems to match yours
[image]

When I run the log command, it also seems to be downloading the testnet3 blocks:

$ sudo tail -f /home/bitcoin/.bitcoin/testnet3/debug.log
[image]

But when I follow the next step, I get the following error:
$ bitcoin-cli getblockchaininfo
> error: couldn't connect to server: unknown (code -1)
(make sure server is running and you are connecting to the correct RPC port)

From my limited experieince of running a node, this is usually because I have yet to run the command
# bitcoind -daemon

However, I can see the blocks downloading due to the service we created. Any ideas as to why I am getting this error? This is all from the admin user created through the tutorial.

Stadicus, 2018-04-19:
The `bitcoind.service` takes care of starting the `bitcoind`, with the -daemon option specified in the configuration file. So that's not the issue, you never need to manually touch `bitcoind`, only `bitcoin-cli`.

It's possible that the system is just too busy to answer during the initial download of the blockchain. As long as it's downloading, you just need to wait for it to finish. 

Another option is that you did not copy the `bitcoin.conf` file into the `/home/admin/.bitcoin` directory of and thus the authentication fails.

nperalez, 2018-04-20:
It seems to be working now. I think the problem was your later comment. That is, I forgot to copy the `bitcoin.conf file`. However, I thought this yesterday and copied it with no luck. I attempted the same thing today after reading your comment. The only thing I can remember doing different, at least I think, is that before I copied the file I stopped the bitcoind.service. After then restarting the pi it seemed to work. Thanks again!

closed 2018-04-20 by nperalez