raspibolt/raspibolt #63: raspibolt_balance

robclark56, 2018-04-27:
[image]

I found that the way I used RaspiBolt was that I was always checking my channels I had open - with the local & remote balances. This is especially interesting when someone else opens a channel with me.

So I wrote the code below. I am not sure how to do a pull request with only 1 (of the many) files I have been working on. So I have included here. Not elegant ...but...

A few notes:
1) As written there are 2 versions of the channel balance display. Naturally only one is needed. Rather than delete Version 1 - it is in the code ... easy to delete or comment out V1 or V2 as needed.
2) /bin/bash compatible  (As you can guess this is based on the original raspibolt which uses /bin/sh)
3) Has a command line switch (--testnet) which is optional ... and allows it to display either the default or alternative lnd info. (.... I have both running on my RaspiBolt ... so for me the default = mainnet, and alternate = testnet)
 
:-) 
Rob

---

```
#!/bin/bash
# RaspiBolt channel balance display
# /usr/local/bin/raspibolt_balance
# root must be able to execute bitcoin-cli and lncli

# Usage
# $ raspibolt_balance            to display lnd mainnet status
# $ raspibolt_balance --testnet  to display lnd testnet status

# Set Default (mainnet)
lncli='/usr/local/bin/lncli'
lnd_pid=$(systemctl show -p MainPID lnd | awk -F"=" '{print $2}')
chain='main'

# read cli args
for i in "$@"
do
case $i in
    --testnet*)
    lncli="/usr/local/bin/lncli  --rpcserver=localhost:11009"
    lnd_pid=$(systemctl show -p MainPID lnd_testnet | awk -F"=" '{print $2}')
    chain='test'
    shift # past argument=value
    ;;
esac
done

# set colors
color_red='\033[0;31m'
color_green='\033[0;32m'
color_yellow='\033[0;33m'
color_gray='\033[0;37m'

# get LND info
wallet_color="${color_yellow}"
if [ "$lnd_pid" -ne "0" ]; then
 ${lncli} getinfo 2>&1 | grep "Please unlock" >/dev/null
 wallet_unlocked=$?
 if [ "$wallet_unlocked" -eq 0 ] ; then
  wallet_color="${color_red}"
  ln_walletbalance="Locked"
 else
  alias_color="${color_grey}"
  ln_alias="$(${lncli} getinfo | jq -r '.alias')" 2>/dev/null
  ln_walletbalance="$(${lncli} walletbalance | jq -r '.confirmed_balance')" 2>/dev/null
  ln_channelbalance="$(${lncli} channelbalance | jq -r '.balance')" 2>/dev/null
  ln_channels_active="$(${lncli} listchannels --active_only| jq '.[] | length')" 2>/dev/null
  ln_channels_inactive="$(${lncli} listchannels --inactive_only| jq '.[] | length')" 2>/dev/null
  active_remote="$(${lncli} listchannels --active_only | jq -r '.channels |.[] | .remote_balance ' | jq -s 'add')"
  active_local="$(${lncli} listchannels --active_only | jq -r '.channels |.[] | .local_balance ' | jq -s 'add')"
  inactive_remote="$(${lncli} listchannels --inactive_only | jq -r '.channels |.[] | .remote_balance ' | jq -s 'add')"
  active_fees="$(${lncli} listchannels --active_only | jq -r '.channels |.[] | .commit_fee ' | jq -s 'add')"
  inactive_fees="$(${lncli} listchannels --inactive_only | jq -r '.channels |.[] | .commit_fee ' | jq -s 'add')"
  inactive_local="$(${lncli} listchannels --inactive_only | jq -r '.channels |.[] | .local_balance ' | jq -s 'add')"
 if [ "${active_local}" = 'null' ];then active_local=0;fi
 if [ "${active_remote}" = 'null' ];then active_remote=0;fi
 if [ "${inactive_local}" = 'null' ];then inactive_local=0;fi
 if [ "${active_fees}" = 'null' ];then active_fees=0;fi
 if [ "${inactive_fees}" = 'null' ];then inactive_fees=0;fi
 if [ "${inactive_remote}" = 'null' ];then inactive_remote=0;fi
 if [ "${ln_walletbalance}" = 'null' ];then ln_walletbalance=0;fi
 if [ "${ln_walletbalance}" = 'Locked' ];then ln_walletbalance=0;fi
 total_local=$(( ${ln_walletbalance} + ${active_local} + ${inactive_local} ))
 total_remote=$(( ${active_remote} + ${inactive_remote} ))
 total_fees=$(( ${active_fees} + ${inactive_fees} ))
 fi
else
 wallet_color="${color_red}"
 ln_walletbalance="Not Running"
fi

#### VERSION 1  ####
margin='                    '
printf "
${margin}${color_red}|    Inactive Channels    ${color_yellow}|   %7s   |${color_green}     Active Channels     |
${margin}${color_red}|Remote              Local${color_yellow}|    Wallet   |${color_green}Local              Remote|
${margin}${color_red}|%-12s %12s${color_yellow}|${wallet_color}%12s ${color_yellow}|${color_green}%-12s %12s|
" \
"${chain}net" \
"${inactive_remote}" "${inactive_local}"  "${ln_walletbalance}" "${active_local}" "${active_remote}"

#### VERSION 2  ####
margin=''
printf "
${margin}${color_yellow}%-21s${color_gray}|       Local|      Remote|Commitment Fees|
${margin}${color_gray}%-21s|${color_yellow}%12s${color_gray}|%12s|%15s|
${margin}${color_gray}%-21s|${color_green}%12s${color_gray}|${color_green}%12s|${color_red}%15s${color_gray}|
${margin}${color_gray}%-21s|${color_red}%12s${color_gray}|${color_red}%12s${color_gray}|${color_red}%15s${color_gray}|
${margin}${color_gray}%-21s|%12s|%12s|${color_red}%15s${color_gray}|
" \
"${chain}net (sat)" \
"Wallet" "${ln_walletbalance}" "" "" \
"Active Channels (${ln_channels_active})" "${active_local}" "${active_remote}" "${active_fees}" \
"Inactive Channels (${ln_channels_inactive})" "${inactive_local}"  "${inactive_remote}" "${inactive_fees}" \
"Total" "${total_local}" "${total_remote}" "${total_fees}"

echo "$(tput -T xterm sgr0)"

```

Stadicus, 2018-04-28:
Wow, that's really cool. Will check it out, looks like a great addition to the Bonus section!

Stadicus, 2018-04-28:
Btw: you can create a new "branch" in your repository, commit this one file to it, and then create a PR for this branch.

Stadicus, 2018-05-25:
Included the script here:  
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_67_additional-scripts.md

closed 2018-05-25 by Stadicus