raspibolt/raspibolt #64: raspibolt_channels

robclark56, 2018-04-29:
[ Still can't workout how to create a branch with ONLY this change in it.]
[image]

This is a candidate for the Bonus section

Rob

---

```
#!/bin/bash
###############################################
#  /usr/local/bin/raspibolt_channels
#
#  Shows details of current channels
#
###############################################

# Usage
# $ raspibolt_channels            to display lnd mainnet channels
# $ raspibolt_channels --testnet  to display lnd testnet channels

# Set Default (mainnet)
lncli='/usr/local/bin/lncli'
lnd_pid=$(systemctl show -p MainPID lnd | awk -F"=" '{print $2}')
chain='main'

# read cli args
for i in "$@"
do
case $i in
    --testnet*)
    lncli="/usr/local/bin/lncli  --rpcserver=localhost:11009"
    lnd_pid=$(systemctl show -p MainPID lnd_testnet | awk -F"=" '{print $2}')
    chain='test'
    shift # past argument=value
    ;;
esac
done

if [ "$lnd_pid" -eq "0" ]; then
 echo lnd not runnning.
 exit
fi

# set colors
color_red='\033[0;31m'
color_green='\033[0;32m'
color_yellow='\033[0;33m'
color_gray='\033[0;37m'

# gather values
a_active=( $(${lncli} listchannels | jq -r ' .channels[].active'))
a_remote_pubkey=( $(${lncli} listchannels | jq -r ' .channels[].remote_pubkey'))
a_capacity=( $(${lncli} listchannels | jq -r ' .channels[].capacity'))
a_local_balance=( $(${lncli} listchannels | jq -r ' .channels[].local_balance'))
a_remote_balance=( $(${lncli} listchannels | jq -r ' .channels[].remote_balance'))
a_commit_fee=( $(${lncli} listchannels | jq -r ' .channels[].commit_fee'))
a_channel_point=( $(${lncli} listchannels | jq -r ' .channels[].channel_point'))

total=${#a_active[*]}
total_capacity=0
total_fee=0
total_local=0
total_remote=0

#display
printf "${color_yellow}%-7s%60s %11s\n" "${chain}net" 'Commit ------- Balance ---------' '--- Fee ----'
printf "%-21s %12s %5s %12s %12s %6s %5s\n" 'Alias or Pubkey' 'Capacity' 'Fee' 'Local' 'Remote' 'Base' 'PerMil'
horiz_line="-------------------- ------------- ------ ------------ ------------ ----- ------"
echo $horiz_line
for (( i=0; i<=$(( $total -1 )); i++ ));do
 if [ ${a_active[$i]} == 'true' ];  then
  color_line=${color_gray}
 else
  color_line=${color_red}
 fi
 alias=$(${lncli} getnodeinfo ${a_remote_pubkey[$i]} | jq -r .node.alias)
 if [ "${alias}" == "" ] ; then
  alias_short=$(echo ${a_remote_pubkey[$i]} | cut -c-17)...
 else
  alias_short=$(echo ${alias} | cut -c-20)
 fi
 active_short=$(echo ${a_active[$i]} | cut -c1)
 # get fee report details
 base_fee_msat=$(${lncli} feereport | jq -r ".channel_fees[] | select(.channel_point | test(\"${a_channel_point[$i]}\")) | .base_fee_msat")
 fee_per_mil=$(${lncli} feereport | jq -r ".channel_fees[] | select(.channel_point | 
 test(\"${a_channel_point[$i]}\")) | .fee_per_mil")
 # Display line 
 printf "${color_line}%-21s %12s %6s %12s %12s %5s %6s\n" \
        "${alias_short}" "${a_capacity[$i]}" "${a_commit_fee[$i]}" "${a_local_balance[$i]}" \
        "${a_remote_balance[$i]}" "${base_fee_msat}" "${fee_per_mil}"
 total_capacity=$(( ${total_capacity} + ${a_capacity[$i]} ))
 total_fee=$(( ${total_fee} + ${a_commit_fee[$i]} ))
 total_local=$(( ${total_local} + ${a_local_balance[$i]} ))
 total_remote=$(( ${total_remote} + ${a_remote_balance[$i]} ))
done
printf "${color_yellow}%s\n" "${horiz_line}"
printf "Totals%14s %13s %6s %12s %12s" "${total} ch" "${total_capacity}" "${total_fee}" "${total_local}" "${total_remote}"
echo "$(tput -T xterm sgr0)"
```

Stadicus, 2018-05-03:
Hi Rob, I love it. However, I don't have the time to update the guide atm. Could you try to create a full PR, incl. updated Bonus section table-of-content and the whole framing of this new page? 

If not, I'll try to include it sometime in June, but it's a pitty to wait that long... :-)

Stadicus, 2018-05-25:
I tested your scripts and this one works very well without changes, except for the layout when channels use non-latin characters. printf does not seem to handle this correctly. Maybe a cleansing (UTF > ASCII) is needed, or an alternate way of alignment?

[image]

robclark56, 2018-05-25:
OK ... will see what I can do after
1) Returning to AU (I have no direct access to the RaspiBolt)
2) Working out how to log in to my RaspiBolt. Again my testnet lnd is in a state where I can't login as lnd listchannels hangs .... and that means MOTD hangs. I am told next release has a fix for the lnd bug.

Stadicus, 2018-05-26:
Great. Regarding the MOTD hanging, this is why I set up the guide to still allow "root" to log in (without MOTD), you could try that. And I guess the inclusion of a timeout in the MOTD would be helpful  
http://www.fibrevillage.com/sysadmin/236-how-to-timeout-a-command-in-linux-shell

robclark56, 2018-06-01:
Back ... and have a simple fix for the formatting issue.

change:
 # Display line 
 printf "${color_line}%-21s %12s %6s %12s %12s %5s %6s\n" \
        "${alias_short}" "${a_capacity[$i]}" "${a_commit_fee[$i]}" "${a_local_balance[$i]}" \
        "${a_remote_balance[$i]}" "${base_fee_msat}" "${fee_per_mil}"

to:
 # Display line 
 printf "${color_line}%-21s %12s %6s %12s %12s %5s %6s\r%-21s\n" \
        "" "${a_capacity[$i]}" "${a_commit_fee[$i]}" "${a_local_balance[$i]}" \
        "${a_remote_balance[$i]}" "${base_fee_msat}" "${fee_per_mil}"  "${alias_short}"

The trick is to print ${alias_short} last and precede with "\r"

Stadicus, 2018-06-05:
Thanks for the great fix, both scripts are now included:  
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_67_additional-scripts.md

closed 2018-06-05 by Stadicus