raspibolt/raspibolt #65: bitcoind.service switch from activating (auto-restart) to active (running) and so on

Amiga500, 2018-05-01:
I copied blockchain (checked from bitcoin core on pc) into external hd connected to raspberry pi, but now i have this:

-----------------------------------------------------------------------------------------------------------------------
pi@raspberrypi:~ $ systemctl status bitcoind.service
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2018-05-01 03:59:35 CEST; 30s ago
  Process: 1067 ExecStart=/usr/local/bin/bitcoind -pid=/home/pi/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 1068 (code=exited, status=1/FAILURE)

May 01 03:59:35 raspberrypi systemd[1]: bitcoind.service: Main process exited, code=exited, status=1/FAILURE
May 01 03:59:35 raspberrypi systemd[1]: bitcoind.service: Unit entered failed state.
May 01 03:59:35 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
May 01 04:00:05 raspberrypi systemd[1]: bitcoind.service: Service hold-off time over, scheduling restart.
May 01 04:00:05 raspberrypi systemd[1]: Stopped Bitcoin daemon.
May 01 04:00:05 raspberrypi systemd[1]: Starting Bitcoin daemon...
-------------------------------------------------------------------------------------------------------------------
pi@raspberrypi:~ $ systemctl status bitcoind.service
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-05-01 04:00:06 CEST; 4s ago
  Process: 1244 ExecStart=/usr/local/bin/bitcoind -pid=/home/pi/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 1245 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─1245 /usr/local/bin/bitcoind -pid=/home/pi/.bitcoin/bitcoind.pid

May 01 04:00:05 raspberrypi systemd[1]: Starting Bitcoin daemon...
May 01 04:00:06 raspberrypi systemd[1]: Started Bitcoin daemon.
-------------------------------------------------------------------------------------------------------------------

and so on.. it switched from activating to active (running state)

this is the bitcoin.conf:
-------------------------------------------------------------------------------------------------------------------
# RaspiBolt LND Mainnet: bitcoind configuration
# /home/pi/.bitcoin/bitcoin.conf

# remove the following line to enable Bitcoin mainnet
#testnet=1

# Bitcoind options
server=1
daemon=1
txindex=1
disablewallet=1

# Connection settings
rpcuser=pi
rpcpassword=xxx
zmqpubrawblock=tcp://127.0.0.1:29000
zmqpubrawtx=tcp://127.0.0.1:29000

# Raspberry Pi optimizations
dbcache=100
maxorphantx=10
maxmempool=50
maxconnections=40
maxuploadtarget=5000
-------------------------------------------------------------------------------------------------------------------

What's wrong? Thanks in advance

Amiga500, 2018-05-01:
2018-05-01 02:33:24 Bitcoin Core version v0.16.0 (release build)
2018-05-01 02:33:24 InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2018-05-01 02:33:24 Assuming ancestors of block 0000000000000000005214481d2d96f898e3d5416e43359c145944a909d242e0 have valid signatures.
2018-05-01 02:33:24 Setting nMinimumChainWork=000000000000000000000000000000000000000000f91c579d57cad4bc5278cc
2018-05-01 02:33:24 Using the 'standard' SHA256 implementation
2018-05-01 02:33:24 Default data directory /home/pi/.bitcoin
2018-05-01 02:33:24 Using data directory /home/pi/.bitcoin
2018-05-01 02:33:24 Using config file /home/pi/.bitcoin/bitcoin.conf
2018-05-01 02:33:24 Using at most 40 automatic connections (1024 file descriptors available)
2018-05-01 02:33:24 Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2018-05-01 02:33:24 Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2018-05-01 02:33:24 Using 4 threads for script verification
2018-05-01 02:33:24 scheduler thread start
2018-05-01 02:33:24 HTTP: creating work queue of depth 16
2018-05-01 02:33:24 Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcuser for rpcauth auth generation.
2018-05-01 02:33:24 HTTP: starting 4 worker threads
2018-05-01 02:33:24 Cache configuration:
2018-05-01 02:33:24 * Using 12.5MiB for block index database
2018-05-01 02:33:24 * Using 8.0MiB for chain state database
2018-05-01 02:33:24 * Using 79.5MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2018-05-01 02:33:24 init message: Loading block index...
2018-05-01 02:33:24 Opening LevelDB in /home/pi/.bitcoin/blocks/index
2018-05-01 02:33:24 Opened LevelDB successfully
2018-05-01 02:33:24 Using obfuscation key for /home/pi/.bitcoin/blocks/index: 0000000000000000
2018-05-01 02:33:47 LoadBlockIndexDB: last block file = 1248
2018-05-01 02:33:47 LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=50, size=38775439, heights=520502...520615, time=2018-04-30...2018-04-30)
2018-05-01 02:33:47 Checking all blk files are present...
2018-05-01 02:33:48 LoadBlockIndexDB: transaction index disabled
2018-05-01 02:33:48 : You need to rebuild the database using -reindex to change -txindex.
Please restart with -reindex or -reindex-chainstate to recover.
2018-05-01 02:33:48 Aborted block database rebuild. Exiting.
2018-05-01 02:33:48 Shutdown: In progress...
2018-05-01 02:33:48 scheduler thread interrupt
2018-05-01 02:33:48 Shutdown: done
2018-05-01 02:34:19

2018-05-01 02:34:19 Bitcoin Core version v0.16.0 (release build)
2018-05-01 02:34:19 InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2018-05-01 02:34:19 Assuming ancestors of block 0000000000000000005214481d2d96f898e3d5416e43359c145944a909d242e0 have valid signatures.
2018-05-01 02:34:19 Setting nMinimumChainWork=000000000000000000000000000000000000000000f91c579d57cad4bc5278cc
2018-05-01 02:34:19 Using the 'standard' SHA256 implementation
2018-05-01 02:34:19 Default data directory /home/pi/.bitcoin
2018-05-01 02:34:19 Using data directory /home/pi/.bitcoin
2018-05-01 02:34:19 Using config file /home/pi/.bitcoin/bitcoin.conf
2018-05-01 02:34:19 Using at most 40 automatic connections (1024 file descriptors available)
2018-05-01 02:34:19 Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2018-05-01 02:34:19 Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2018-05-01 02:34:19 Using 4 threads for script verification
2018-05-01 02:34:19 scheduler thread start
2018-05-01 02:34:19 HTTP: creating work queue of depth 16
2018-05-01 02:34:19 Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcuser for rpcauth auth generation.
2018-05-01 02:34:19 HTTP: starting 4 worker threads
2018-05-01 02:34:19 Cache configuration:
2018-05-01 02:34:19 * Using 12.5MiB for block index database
2018-05-01 02:34:19 * Using 8.0MiB for chain state database
2018-05-01 02:34:19 * Using 79.5MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2018-05-01 02:34:19 init message: Loading block index...
2018-05-01 02:34:19 Opening LevelDB in /home/pi/.bitcoin/blocks/index
2018-05-01 02:34:19 Opened LevelDB successfully
2018-05-01 02:34:19 Using obfuscation key for /home/pi/.bitcoin/blocks/index: 0000000000000000
2018-05-01 02:34:43 LoadBlockIndexDB: last block file = 1248
2018-05-01 02:34:43 LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=50, size=38775439, heights=520502...520615, time=2018-04-30...2018-04-30)
2018-05-01 02:34:43 Checking all blk files are present...
2018-05-01 02:34:43 LoadBlockIndexDB: transaction index disabled
2018-05-01 02:34:43 : You need to rebuild the database using -reindex to change -txindex.
Please restart with -reindex or -reindex-chainstate to recover.
2018-05-01 02:34:43 Aborted block database rebuild. Exiting.
2018-05-01 02:34:43 Shutdown: In progress...
2018-05-01 02:34:43 scheduler thread interrupt
2018-05-01 02:34:43 Shutdown: done

Amiga500, 2018-05-01:
if i use bitcoin-cli -reindex command.. it will take months to finish. There is another command that makes it more fast? I read on twitter that some user after some minutes that copy the blockchain from windows to the raspberry done it perfectly working.

Stadicus, 2018-05-01:
It seems you did not download the Bitcoin blockchain with `txindex=1` option set on your regular computer. You are right, reindexing on the Pi is going to take ages. 

Make sure you download the blockchain like described here: https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_10_preparations.md#installing-bitcoin-core

The important part is the `txindex=1`

Amiga500, 2018-05-01:
Thanks for now Stadicus, i jumped that part of the guide as i have it all downloaded.. so i thinked it was good as default.. i'll let you know when done

Amiga500, 2018-05-05:
Ok, now it works! But i suggest to add this command to the guide, as at the reload the hd wasn't recognized anymore (problem with mount):

sudo fsck -yv /dev/sda1

closed 2018-05-05 by Amiga500