raspibolt/raspibolt #67: Error on Electrum guysd

Sowiedu, 2018-05-01:


Stadicus, 2018-05-02:
Not sure what causes this, might be a recent commit to Electrum Personal Server? 

Try to clone the latest release (tagged "v0.1.1") instead of the "master" branche
https://github.com/chris-belcher/electrum-personal-server/releases
https://stackoverflow.com/questions/791959/download-a-specific-tag-with-git

Sowiedu, 2018-05-03:
Alright, will do tonight. Thanks for the pointer! I will report back.

Sowiedu, 2018-05-08:
Alright, I did what you said and still get the same error. I noticed my mpk starts with a z instead of an x like in the sample. Is that an issue? Also I didn't put in any watch-only-adresses, but that shouldn't matter, right? only my mpk.

Stadicus, 2018-05-24:
I just reinstalled the EPS last week from the master branch, it worked fine. If you still run into issues, I would recommend opening an issue on the https://github.com/chris-belcher/electrum-personal-server repository.

closed 2018-05-24 by Stadicus