raspibolt/raspibolt #68: create chat / support forum for these guides (and others?)

LilChange, 2018-05-02:
what do you think about creating or linking to a Riot, Mattermost, or IRC type chat room whereby ppl can get real time chat support with setting up their RaspiBolt / Thundroid / other possible future similar offerings?

this is not remotely a criticism of @Stadicus wonderful guides. thank you so much for your outstanding creation here :smile: my thinking is just that real time chat can provide welcoming help to curious newcomers and help build community in a way that issue trackers can not.

i would definitely be up for idling in such a chan and helping out if theres interest.

Stadicus, 2018-05-03:
How about this? 
https://raspibolt.slack.com

Not sure how much time I have a the moment for live support, I think this depends heavily on the help of expirienced RaspiBolters.

Sowiedu, 2018-05-03:
Phenomenal idea. In the spirit of enabling regular joes to run btc/lnd nodes even easier then! One could also start an irc channel maybe?

Stadicus, 2018-05-24:
Not sure if instant support works that well, I already struggle to keep up with clearly written issues here. Will put this on hold for the moment.

closed 2018-05-24 by Stadicus