raspibolt/raspibolt #70: lncli newaddress np2wkh - grpc error

Amiga500, 2018-05-05:
I followed the guide in order to add lnd for mainnet (at blockchain level it's all working nice) but after created lncli..

admin@raspberrypi:~ $ lncli newaddress np2wkh
[lncli] Wallet is encrypted. Please unlock using 'lncli unlock', or set password using 'lncli create' if this is the first time starting lnd.
admin@raspberrypi:~ $ lncli unlock
Input wallet password:

lnd successfully unlocked!
admin@raspberrypi:~ $ lncli newaddress np2wkh
[lncli] rpc error: code = Unavailable desc = grpc: the connection is unavailable
admin@raspberrypi:~ $ sudo journalctl -f -u lnd
-- Logs begin at Thu 2016-11-03 18:16:42 CET. --
May 05 14:08:30 raspberrypi lnd[2263]: 2018-05-05 14:08:30.303 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
May 05 14:08:30 raspberrypi lnd[2263]: 2018-05-05 14:08:30.304 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
May 05 14:08:30 raspberrypi lnd[2263]: 2018-05-05 14:08:30.304 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create wallet, or `lncli unlock` to unlock already created wallet.
May 05 14:09:12 raspberrypi lnd[2263]: 2018-05-05 14:09:12.467 [INF] LNWL: Opened wallet
May 05 14:09:12 raspberrypi lnd[2263]: 2018-05-05 14:09:12.899 [INF] LTND: Primary chain is set to: bitcoin
May 05 14:09:12 raspberrypi lnd[2263]: 2018-05-05 14:09:12.900 [INF] LTND: Initializing bitcoind backed fee estimator
May 05 14:09:19 raspberrypi lnd[2263]: 2018-05-05 14:09:19.136 [INF] LNWL: Opened wallet
May 05 14:09:19 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
May 05 14:09:19 raspberrypi systemd[1]: lnd.service: Unit entered failed state.
May 05 14:09:19 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.

Amiga500, 2018-05-07:
I opened this issue on LND: https://github.com/lightningnetwork/lnd/issues/1192

Amiga500, 2018-05-08:
Resolved.. if someone will have same problem.. i have changed the password on bitcoin.conf file with another one.. saved and.. all works like a charm :)

Amiga500, 2018-05-24:
No, i didn't use special characters, it was a group of words with capital letter separated with spaces. I think that it wasn't related to the password used.. but only the saving of the password adjust something.

Stadicus, 2018-05-26:
Thanks for the update. I've never used spaces in a pw before. Maybe that was the issue? Anyway, I made the note about not using special characters a bit more bold, maybe that helps.

closed 2018-05-26 by Stadicus