raspibolt/raspibolt #71: hint on bootstrapped blockchain data

s72785, 2018-05-07:
very helpful in case of testing/frequent setups of bitcoin clients…

there used to be an updated file available via bittorrent; this you can create yourself: https://bitcoin.stackexchange.com/a/42916/49086 and then utilize.

s72785, 2018-05-07:
true, this is mainly useful if you e.g. bootstrap several (or the same again) setups as e.g. POS. if download matters depends heavily on your connection/bandwidth, don't expect everybody to have that – not speaking of free volume. and quite some people will be happy to not use their limits for setting up their VPS again.

Stadicus, 2018-05-24:
I am really looking forward to running local Neutrino server https://github.com/lightninglabs/neutrino to deliver the Bitcoin blockchain info to all my various nodes... :-)

closed 2018-05-24 by Stadicus