raspibolt/raspibolt #76: REQUEST: add the procedure to securely upgrade BITCOIN and LND bin

Amiga500, 2018-06-03:
Why not add in the guide the procedure to upgrade LND and BITCOIN bin as now are both surpassed?

I hope these are the correct steps (all from ADMIN user):

$ sudo systemctl stop lnd
$ sudo systemctl stop bitcoind

then download releases:

cd /home/admin/download
wget https://bitcoin.org/bin/bitcoin-core-0.16.1/bitcoin-0.16.1rc1-arm-linux-gnueabihf.tar.gz
tar -xvf bitcoin-0.16.1rc1-arm-linux-gnueabihf.tar.gz
sudo install -m 0755 -o root -g root -t /usr/local/bin bitcoin-0.16.1/bin/*

cd /home/admin/download
$ wget https://github.com/lightningnetwork/lnd/releases/download/v0.4.2-beta/lnd-linux-arm-v0.4.2-beta.tar.gz
tar -xzf lnd-linux-arm-v0.4.2-beta.tar.gz
sudo install -m 0755 -o root -g root -t /usr/local/bin lnd-linux-arm-v0.4.2-beta/*

$ sudo systemctl start lnd
$ sudo systemctl start bitcoind

--------------------------------------------
bitcoin@raspberrypi:~ $ bitcoin-cli -version
Bitcoin Core RPC client version v0.16.1rc1
bitcoin@raspberrypi:~ $ lncli -v
lncli version 0.4.2 commit=7cf5ebe2650b6798182e10be198c7ffc1f1d6e19
--------------------------------------------

Stadicus, 2018-06-03:
Your procedure should ve correct, no specual steps necessary. 

When upgrading from lnd 0.4.0, ghe macaroon files need to be recreated. see https://github.com/Stadicus/guides/issues/57

Also, I would not recommend running a Release Candidate (bitcoind rc1) version for real, only for testing.

Amiga500, 2018-06-03:
Thanks for the reply, luckily my first version was 0.4.1. 

I know but.. as the bitcoin core crew is the best and 0.16.1 and follow is the branch of only bug corrections and no new features.. i trust more ;)