raspibolt/raspibolt #79: (not a bug) on raspibolt: Public Not reachable

Amiga500, 2018-06-13:
On raspibolt script from few days i have this:

฿itcoin (mainnet)
Sync    OK 
Public  Not reachable

In past, Public was OK, but now it's on "Not reachable" state. 
I didn't change anything.. what could be happened? 
The node however is working good, i can see it online from eclair wallet as i use that node.

Cees2daBees, 2018-06-13:
+1
I am seeing the same issue. Not sure what is causing it, but I’ve been ignoring it since it seems my node is working as expected. Would be interested as well to see what’s causing this status.

Stadicus, 2018-06-13:
The script uses the API of https://bitnodes.earn.com/ to check if a bitcoin client can be detected on your public ip address and port 8333. You can also check manually on that website. Could you please check and post here whether your public IP is correct and our Bitcoin node is detected?

Cees2daBees, 2018-06-14:
checked my node on https://bitnodes.earn.com/ and result returned:

24.212.167.253:8333 is unreachable.

ip address looks right.. i double checked here: https://whatismyipaddress.com/

waffi121, 2018-06-14:
I have the same problem recently. on my IP the port is open, checked on www.yougetsignal.com/

in the debuglog:

`2018-06-14 10:31:24 UpdateTip: new best=00000000000000000037ffc9b036061dba286fb2d87801ab02e7275588811492 height=527414 version=0x20000000 log2_work=89.02653 tx=322316959 date='2018-06-14 10:16:25' progress=0.999990 cache=46.5MiB(416052txo) warning='3 of last 100 blocks have unexpected version'
`

maybe something with the warning 3?

Stadicus, 2018-06-14:
Could be multiple issues, including with the API of earn.com. As long as `bitcoind` is getting the latest blocks, everything's fine. If others can connect to your node, even better.

Things to check:
* public ip address
* port forwarding in the router (port 8333)
* firewall settings on RaspiBolt (ufw)
* bitcoind config (non-standard port?)

Intersting issue, but non-critical as long as bitcoind is in sync with the Bitcoin network.

waffi121, 2018-06-14:
have all points through. at https://bitnodes.earn.com/ I am connected again but the raspberry is still: Not reachable.

I think they just have problems with the API. Funny the same 3 have the same problem.

Amiga500, 2018-06-14:
Yesterday evening for few hours the state Public change to "YES" and later again on Not reachable.. very strange.. I didn't change anything..

waffi121, 2018-06-15:
is good again! public: yes :) probably really something with the API.

Stadicus, 2018-06-16:
Looks like the API throttles traffic from time to time:
https://github.com/Stadicus/guides/issues/81

Abcesso, 2018-06-19:
I also did have some issue about Public Not reachable who did start a few days ago.

But the last days i did have public Not reachable, and i did close all my channels. The satoshi i did have did never come back to my wallet when i had this issue. But today i did have some time Public os OK. And my satoshi id come back to wallet after the block is finnish. And now i have Public Not reachable again and i cant open a new channel. Any suggest who can fix this issue?

Thanks

Stadicus, 2018-06-19:
Two things first:
* The script showing "Public Not reachable" uses the information of an external service that tries to check the bitcoind node. This is not necessarily reliable information, as we've seen that this API sometimes throttles traffic.
* The service only checks the `bitcoind` node and has no connection to `lnd`

To check your public visibility, I'd recommend going through these checks:

* `bitcoind` first
  * Make sure that the daemon is running: `sudo systemctl status bitcoind`
  * Check if it is synced to the blockchain and connected to peers: `bitcoin-cli -getinfo`
  * compare "blocks" with the latest block on https://www.smartbit.com.au (or https://testnet.smartbit.com.au for Bitcoin testnet)
  * "connections" should be > 0
  * Check firewall to make sure port 8333 (or 18333 on testnet) is open: `sudo ufw status`
    (`8333                       ALLOW       Anywhere`)
  * Use https://bitnodes.earn.com to check if it sees your Bitcoin node
  * If not successful, check your router port forwarding settings (it must point port 8333 to the internal ip address for your RaspiBolt)

* If successful, go on to `lnd`
  * Check if daemon is running, and whitch ip is announced: `sudo systemctl status lnd`
    eg. `-28100 /usr/local/bin/lnd --externalip=91.190.20.11`
  * Check if its "synced_to_chain" and has "num_active_channels": `lncli getinfo`
  * If in doubt, check if you can connect to a node from https://1ml.com, you should get empty brackets `{   }` as return
    `lncli connect 02a45def9ae014fdd2603dd7033d157faa3a55a72b06a63ae22ef46d9fafdc6e8d@78.42.161.43:9735`
    * Check firewall to make sure port 9735 is open: `sudo ufw status`
(`9735 ALLOW Anywhere`)
    * Make sure to check your router port forwarding for port 9735

Let me know if these step help or if you can think of an additional step to make it as bulletproof as possible. Will incorporate into FAQ once its final.

Amiga500, 2018-06-19:
All check passed.. but always Public  Not reachable

Cees2daBees, 2018-06-20:
Also went through all these steps and they check out.. and still Public Not reachable

Stadicus, 2018-06-21:
Hm, started having this issue as well, although `bitcoind` and `lnd` are fully operational. I think the API method should be replcaed, I'll try the method recommended by @robclark56 
https://github.com/Stadicus/guides/issues/81#issuecomment-397921941

Amiga500, 2018-06-21:
I have tried the method.. but it seems it doesn't work.. even with reload of the rpi..

robclark56, 2018-06-21:
@Amiga500 

My bitcoind is using this Public IP:Port (as seen on the raspibolt report)

 124.148.74.13:8333

And this is what I see:

admin ~  ฿  timeout 2s nc -z 124.148.74.13 8333;echo $?
0

What do you see if you do the same (replacing the IP,Port with your numbers)?

Amiga500, 2018-06-21:
Hi @robclark56 

bitcoin@raspberrypi:~ $ timeout 2s nc -z 2.238.78.69 8333; echo $
$
bitcoin@raspberrypi:~ $

robclark56, 2018-06-22:
@Amiga500  ... you missed the '?' on the end of the command. It is

admin ~ ฿ timeout 2s nc -z 124.148.74.13 8333;echo $?

robclark56, 2018-06-23:
@Amiga500 
The '0' is good. It means your node is public.

See this as a suggestion and let us know if it works:
https://github.com/Stadicus/guides/issues/81
(4th message in that thread)

Amiga500, 2018-06-23:
@robclark56 I did change the code days ago as that thread but nothing changed :(

Amiga500, 2018-06-24:
@robclark56
Yes i changed it so, but retyping raspibolt command for 4 - 5 times.. it return to Not Reachable.. i don't know why...

[image]

Stadicus, 2018-07-12:
Updated welcome script in PR https://github.com/Stadicus/guides/pull/121

closed 2018-07-12 by Stadicus

ejose19, 2018-12-20:
I suggest to leave both options working, or leave one of them commented so the user can easily use whatever suits he best.

I say this because the netcat solution doesn't work for me due missing NAT Loopback functionability:

"Many DSL routers/modems prevent loopback connections as a security feature. This means that a machine on your local network (e.g. behind your DSL router/modem) cannot connect to a forward facing IP address (such as 199.149.252.44) of a machine that it also on your local network. Connecting to the local IP address (such as 192.168.2.40) of that same machine works fine."

Since I guess not everyone has a router with NAT Loopback, some users will be fine with bitnodes solution (it did for me, I modified the script to use bitnodes only and now everything is good)

53645714n, 2019-10-08:
> I suggest to leave both options working, or leave one of them commented so the user can easily use whatever suits he best.
>
> I say this because the netcat solution doesn't work for me due missing NAT [...]

I did the same and left it commented in, will do a PR in the near future where I will leave this + some version info & update checking.

53645714n, 2019-10-09:
> I suggest to leave both options working, or leave one of them commented so the user can easily use whatever suits he best.
> I say this because the netcat solution doesn't work for me due missing NAT [...]
 
I did the same and left it commented in, will do a PR in the near future where I will leave this + some version info & update checking.

jiyoon-koo, 2022-05-02:
What should I do if `timeout 2s nc -z {my_ip_addr} 8333; echo $?` returns 1? 

I was thinking I would run this command, but I want to make sure that's the right answer before poking this hole open on the firewall.
`ufw allow 8333    comment 'allow other BTC nodes'`