raspibolt/raspibolt #81: /usr/local/bin/raspibolt and https://bitnodes.earn.com/api/v1/nodes/me-${public_port}

robclark56, 2018-06-14:
Noticed today that the raspibolt script was saying: 
Public  Not reachable

Found out why.
https://bitnodes.earn.com/api/v1/nodes/me-${public_port} was returning:

{"detail":"Request was throttled."}

Considering that raspibolt runs rarely (only on login), the service can not be throttling on requests/hour.  Maybe the total number of requests from my IP address is the problem.

Looking for an alternative service ... but no luck just yet.

Stadicus, 2018-06-16:
Good catch, might be the cause for some of the issues raised here. Do you know of a different method to check the public availability of the Bitcoin node?

robclark56, 2018-06-16:
Sorry no.

Tried without success to fine an alternative service.

Stadicus, 2018-07-12:
Hi @robclark56, I tried to figure out how this should work. As long as I use my own ip address, netcat always return no value. Which means 'you are public', I guess?

[image]

Am I missing something?

robclark56, 2018-07-12:
nc -z IP PORT returns:

- 0 if success (ie PORT on IP is open)
- Non-Zero if failure (ie Port is not open)

Here are my examples (8333 is open, 12345 is not)
```
admin ~  ฿  nc -z  106.68.62.191 8333
admin ~  ฿  echo $?
0
admin ~  ฿  nc -z  106.68.62.191 12345
[..... long wait ....]
admin ~  ฿  echo $?
1
```
And here it is verbose
```
admin ~  ฿  nc -vz  106.68.62.191 8333
Connection to 106.68.62.191 8333 port [tcp/*] succeeded!
admin ~  ฿  nc -vz  106.68.62.191 12345
nc: connect to 106.68.62.191 port 12345 (tcp) failed: Connection timed out
```

But you indicate you get 'success' for all ports. That should not happen.

Try the verbose (-zv) option on a closed port.

Stadicus, 2018-07-12:
[image]

Ok, works great! Was not aware that the return value needs to be printed...

Stadicus, 2018-07-12:
Any special reason you went with the `==` operator? I get the following error message:

[image]
`./20-raspibolt-welcome: 117: [: 0: unexpected operator`

When using the normal `=` operator, it works fine.
[image]

robclark56, 2018-07-12:
I think it was because I updated my raspibolt script to bash instead of /bin/sh
#!/bin/bash

In bash you check equality of STRINGS with ==. 
And I converted $public_check to a string as follows: "$public_check"  (added quotes)

But by all means do what works. I am not an expert on shell scripting at all.

:-)

Stadicus, 2018-07-12:
Updated the welcome script, thanks @robclark56!
PR https://github.com/Stadicus/guides/pull/121

closed 2018-07-12 by Stadicus

Janaka-Steph, 2018-10-18:
Hello!
I have some issues checking the public availability of my node (on testnet) with `nc -vz IP PORT`. As my node try to access my public IP address from behind a NAT, he got confused. This is a classic networking problem. 

Reading this blog post [System: Accessing Public IP address from behind NAT](https://www.the-art-of-web.com/system/iptables-nat/) I found what I first thought was a solution, which is adding a rule in iptables to convert my public IP to my local IP.
`$ sudo iptables -A INPUT -p tcp --src 85.220.240.110 --dport 22 -j ACCEPT`

```
$ sudo iptables -L -t nat
Chain OUTPUT (policy ACCEPT)
target       prot    opt          source                 destination
DNAT       all        --            thundroid            85.220.240.110            to:192.168.1.104
```

It seems to work but in fact it is not checking anything because the request never really leave my local network. So not a proper solution.

It seems that this `nc` call works seamlessly for some people but I can't figure out why it doesn't work for me. Any ideas? Thank you.