raspibolt/raspibolt #82: "Get external IP script" not discussed

ulfmartin, 2018-06-15:
We set up a daemon which requires "getpublicip.service", but we did not implement that yet.
There is a missing part in the instructions which is about getting the public IP and saving it
to a temp file in memory. After investigating that you do have the service and the shell script for that purpose somewhere in the repo, I noticed that it does not work for me. I cannot write to /run/. Not even with sudo.
Where those kind of issues the reason why it was (partially) removed from the guide?

Stadicus, 2018-06-16:
Is this the part you are referencing to? Is anything missing?
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_40_lnd.md#public-ip-script

This is somewhat of a hack and will be removed as soon as LND brings full NAT compatibility (should be included in the next release: https://github.com/lightningnetwork/lnd/pull/1109).

ulfmartin, 2018-06-18:
Oh, I now notice what went wrong in the guide.
I the order of steps is a little wrong here.
In the section "Bitcoin", "Autostart bitcoind" the bitcoind.service introduces "Wants=getpublicip.service", before we created that service.
The getpublicip.service and the corresponding sh script is then explained at the beginning of the "Lightning" section. So, I would say it's logical to do it the other way around or make a remark, that we cannot use the bitcoin daemon yet until the following step.

ulfmartin, 2018-06-18:
There was one more thing, actually.
When attempting to write to /run/anyfilename, I would get a "permission denied" error, even when using sudo (e.g. sudo echo sometext > /run/somefile). However, I found that I can fix that issue by using /run/shm/anyfile. So I changed my getpublicip.sh script to output to that folder.
I do not really understand the /run directory, so someone might want to have a second look on that.

Stadicus, 2018-07-07:
I hope to get rid of the getpublicip script asap, as LND supports UPNP / NAT traversal. Will update the guide soon, so this should no longer be an issue. Thanks for your input!

closed 2018-07-07 by Stadicus