raspibolt/raspibolt #83: Will the setup work when behind NAT with no public IP?

bbelo, 2018-06-16:


Stadicus, 2018-06-16:
It's my understanding that every NAT has a public IP address, you can check yours here:
http://www.whatsmyip.org/

However, this ip address can change constantly, and this is one of the challenges not yet addressed in LND but should be included in the next release:
https://github.com/lightningnetwork/lnd/pull/1109

Stadicus, 2018-06-16:
Not sure how exactly the NAT UPnP works, it might be able to set port-forwarding if allowed to. Otherwise you should still be able to connect to other nodes and keeping that communication going, but you won't be able to be a "routing node" and accept incoming channel requests.

Creating a VPN tunnel and being able to set port forwarding on the other end of that tunnel should work in theory. But I never used that before as infrastructure, only with ProtonVPN for adhoc connections.

This is at least my understanding, but I'm on a limb here.

closed 2018-06-16 by Stadicus