raspibolt/raspibolt #85: Cannot send payment to Starblocks

twofrontteeth, 2018-06-17:
I have followed Stadicus's guide (with thanks!) and setup my testnet lnd wallet without issue. Payments to Y'alls work instantly (Fish likes Milk). Unfortunately payments to Starblocks aren't working. I use the same command "lncli sendpayment --pay_req=lnt...." to both sites. I get the following timeout  

 "payment_error": "payment attempt not completed before timeout of 1m0s",
        "payment_preimage": "",
        "payment_route": null

I have had the lightning wallet up and running on testnet for about 72 hours and have nad the same error the whole time. 

TFT

Stadicus, 2018-06-18:
If Y'alls works and you have enough liquidity in your open channels, Starblocks should not be an issue. However, they had connectiviy / liquidity issues in the past, so the problem might not even be on your end.

BTW: you can also just use `lncli payinvoice [INVOICE]` for a faster payment.

If the issue persists, you can ask Acinq directly on their live chat (gitter) on https://gitter.im/ACINQ/eclair, they are very helpful.

twofrontteeth, 2018-06-20:
Thanks for looking at this Stadicus. Unfortunately the same issue remains. I will ask Acinq as suggested. 

I have a further question. My testnet balance doesn't seem to be updating after successful testnet transactions (to Yall's - the articles open in full after the payment). I assume the wallet balance would decrease by the amount/value ie 150. I only raise this on the chance it is related to the other issue rather than PEBKAC. Example:

admin@Raspibolt:~ $ lncli walletbalance
{
    "total_balance": "43999655",
    "confirmed_balance": "43999655",
    "unconfirmed_balance": "0"
}
admin@Raspibolt:~ $ lncli payinvoice lntb1500n1pdj5wfspp5ed2kdpel3qys647hyl80f7le39lyz6pax655qzhu6z5dt2yf67sqdpa2fjkzep6ypfkzar0wd5xj3rfvdjjq6rpwvs8xetkv4exzmpqdejhwgrpv3j8ycqzysedgqyqj5fpm9klrhsg93qy2p7tvmn6jj9e3nnjxsz565npxff77yw3pdaegh76wfawhqmk4x8kdv6kdjf70yusd6qznj58nm45gd4zsqw4vpg4
{
        "payment_error": "",
        "payment_preimage": "234a11054c617203d9f824a6fcd4525bd056892a7a559bcbf78623384c28a18f",
        "payment_route": {
                "total_time_lock": 1326065,
                "total_amt": 150,
                "hops": [
                        {
                                "chan_id": 1457407060675723264,
                                "chan_capacity": 16326338,
                                "amt_to_forward": 150,
                                "expiry": 1326065,
                                "amt_to_forward_msat": 150000
                        }
                ],
                "total_amt_msat": 150000
        }
}
admin@Raspibolt:~ $ lncli listpayments
{
    "payments": [
        {
            "payment_hash": "cb5566873f88090d57d727cef4fbf9897e41683d36a9400afcd0a8d5a889d7a0",
            "value": "150",
            "creation_date": "1529493838",
            "path": [
                "02212d3ec887188b284dbb7b2e6eb40629a6e14fb049673f22d2a0aa05f902090e"
            ],
            "fee": "0",
            "payment_preimage": "234a11054c617203d9f824a6fcd4525bd056892a7a559bcbf78623384c28a18f"
        }
    ]
}
admin@Raspibolt:~ $ lncli walletbalance                                                                 {
    "total_balance": "43999655",
    "confirmed_balance": "43999655",
    "unconfirmed_balance": "0"
}

Again - many thanks for your time and efforts. 

TFT

Cees2daBees, 2018-06-20:
Hi TFT,
the command lncli walletbalance shows your on-chain tBTC balance. 
when making lightning payments they will be drawn from your lightning channel balance and can be checked using:
lncli channelbalance

robclark56, 2018-06-21:
Comment: seeing reports of incompatibility between c-lightning (eclair, starblocks) & lnd. Something to do with compression that is not in the BOLT standards. Not 1000% sure of my facts.

closed 2018-06-24 by Stadicus