raspibolt/raspibolt #88: Bitcoin Core Update to 0.16.1

waffi121, 2018-06-18:
Short question about the update to 0.16.1

BTC and Lightning stop with:

bitcoin-cli stop
sudo systemctl stop lnd

Download with:

 wget https: //bitcoin.org/bin/bitcoin-core-0.16.1 ...
tar -xvf bitcoin-0.16.1-arm-linux-gnueabihf.tar.gz
sudo install -m 0755 -o root -g root -t / usr / local / bin bitcoin-0.16.1 / bin / *

Then reactivate with:
sudo systemctl start lnd

Check the BTC version:

bitcoind --version

Should everything be for the update? :)

Thank you!

Amiga500, 2018-06-18:
Try this (with admin user):

$ sudo systemctl stop lnd
$ sudo systemctl stop bitcoind

cd /home/admin/download
wget https://bitcoin.org/bin/bitcoin-core-0.16.1/bitcoin-0.16.1-arm-linux-gnueabihf.tar.gz
tar -xvf bitcoin-0.16.1-arm-linux-gnueabihf.tar.gz
sudo install -m 0755 -o root -g root -t /usr/local/bin bitcoin-0.16.1/bin/*

$ sudo systemctl start bitcoind
$ sudo systemctl start lnd

waffi121, 2018-06-18:
thank you works fine :)

closed 2018-06-18 by waffi121