raspibolt/raspibolt #89: bitcoin-cli isn't reachable

nipponese, 2018-06-18:
I can't figure out why I can't hit bitcoin-cli with these instructions. The error is:
```
$ bitcoin-cli getblockchaininfo
error: couldn't connect to server: unknown (code -1)
(make sure server is running and you are connecting to the correct RPC port)
```

Entire output as follows:
```
pi@nodeboi:~ $ sudo systemctl status bitcoind.service
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-06-17 22:59:19 PDT; 22min ago
  Process: 416 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitc
 Main PID: 762 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─762 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Jun 17 22:59:18 nodeboi systemd[1]: Starting Bitcoin daemon...
Jun 17 22:59:19 nodeboi systemd[1]: bitcoind.service: PID file /home/bitcoin/.bitcoin/bitcoind.pid not readable (yet?) after start:
Jun 17 22:59:19 nodeboi systemd[1]: Started Bitcoin daemon.

pi@nodeboi:~ $ sudo su bitcoin

bitcoin@nodeboi:/home/pi $ bitcoin-cli getblockchaininfo
error: couldn't connect to server: unknown (code -1)
(make sure server is running and you are connecting to the correct RPC port)

bitcoin@nodeboi:/home/pi $ cat /home/bitcoin/.bitcoin/testnet3/debug.log 
log2_work=53.797313 tx=583312 date='2013-07-18 00:44:35' progress=0.032053 cache=36.4MiB(298262txo)
2018-06-18 05:54:13 UpdateTip: new best=0000000008f55c6a3704919aec00536add6ba10d85788d4fa08d0a0720512eb3 height=96106 version=0x00000002 log2_work=53.797314 tx=583313 date='2013-07-18 01:26:01' progress=0.032053 cache=36.4MiB(298263txo)
2018-06-18 05:54:13 UpdateTip: new best=00000000702feda84df1d93a2e5d39200af30c3ba0fc96c1a5f4be2444de7400 height=96107 version=0x00000002 log2_work=53.797314 tx=583315 date='2013-07-18 01:51:04' progress=0.032054 cache=36.4MiB(298265txo)
2018-06-18 05:54:13 UpdateTip: new best=000000005900f136adbb708e82d94daf1a563da89cceed40962136cbc3271a3d height=96108 version=0x00000002 log2_work=53.797314 tx=583571 date='2013-07-18 02:11:08' progress=0.032068 cache=36.4MiB(298521txo)
...
2018-06-18 06:23:32 UpdateTip: new best=0000000029508914bb1c72a8b3d2670258898204633b1d956c1cd3542ab4488b height=241926 version=0x00000002 log2_work=59.174575 tx=1497887 date='2014-05-10 21:41:52' progress=0.082309 cache=43.2MiB(315442txo)
2018-06-18 06:23:32 UpdateTip: new best=00000000b5bbdcf14dc7c6170ac6e9cad55f96a0af27f4a26c4fbcccbb65d9db height=241927 version=0x00000002 log2_work=59.174575 tx=1497918 date='2014-05-10 22:01:53' progress=0.082311 cache=43.2MiB(315464txo)
2018-06-18 06:23:32 UpdateTip: new best=000000000003f816d1b046edf009145c786814e65a5268206d838da08a56c404 height=241928 version=0x00000002 log2_work=59.174692 tx=1497934 date='2014-05-10 21:37:00' progress=0.082312 cache=43.2MiB(315476txo)
```

nipponese, 2018-06-18:
Looks like there was an issue with the permissions on the debug.log. After I deleted the file for mainnet and testnet, bitcoind regenerated the files and I can use bitcoin-cli now.

closed 2018-06-18 by nipponese

CodingMuziekwijk, 2018-07-12:
@nipponese Could you maybe be a bit more specific on how you fixed it? I am having the same problem and just can't get it to work.