raspibolt/raspibolt #90: How do I upgrade LN if I have 4.1 installed

q0de, 2018-06-20:
You've updated the doc to support the latest LND, but there's no direction to update the previous version to 4.2

Stadicus, 2018-06-20:
Will include this in the FAQ soonish. 

1. If upgrading from version v0.4.0-beta or lower, delete the macaroon files: 
  `sudo rm /home/bitcoin/.lnd/*.macaroon` 
2. Stop LND: `sudo systemctl stop lnd`
3. Download, verify & install LND binaries according to regular guide
4. Start LND: `sudo systemctl start lnd`
5. If you deleted the macaroon files, copy the admin.macaroon file to the admin home directory
  `sudo cp /home/bitcoin/.lnd/admin.macaroon /home/admin/.lnd`
  `sudo chown -R admin:admin /home/admin/.lnd/`

Hope that helps.

Stadicus, 2018-06-22:
See this new FAQ section, thanks to @giulnz!   
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_faq.md#how-to-upgrade-lnd-bin-

closed 2018-06-22 by Stadicus