raspibolt/raspibolt #91: Dashboard to manage own full lightning node.

giulnz, 2018-06-20:
Hi, thanks Stadicus for your guide.
I find lnd.fun, a "Panel for webmasters to manage their full lightning node" but i don't know how to implement and no find info about it.
Someone know a dashboard project for our LN node or how to use that one that i found ?

Stadicus, 2018-06-20:
Yes, http://lnd.fun/ looks gorgeous. Unfortunately, there's no documentation as of now, and no information about the project...

Two better documented projects, that should run without issues:
* https://github.com/altangent/lnd-explorer
* https://github.com/mably/lncli-web

Stadicus, 2018-06-20:
BTW: the iOS & Android app Shango wallet works great to manage basic stuff incl. channels.
https://github.com/neogeno/shango-lightning-wallet

Stadicus, 2018-07-13:
You might want to beta-test this (not published to master yet):
https://github.com/Stadicus/guides/blob/lncli-web/raspibolt/raspibolt_69_lncli-web.md

closed 2018-07-13 by Stadicus