raspibolt/raspibolt #92: Copying The Mainnet blockchain

BitcoinVegan, 2018-06-20:
Hello @stadicus, I don't mean to be a bother but I wanted to know where I can find in the instructions how to copy the mainnet blockchain so I can move to mainnet. 

Thanks for your contribution to the community and support.

Stadicus, 2018-06-20:
It's described here, anything missing?
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_50_mainnet.md

nipponese, 2018-06-20:
@BitcoinVegan If you're on a Mac, may I suggest an alternative using [Rsync](https://rsync.samba.org/). This way you don't need to toggle `PasswordAuthentication` and stay in cli:
```
rsync -r --append -e ssh /Users/noahtsutsui/Library/Application Support/Bitcoin/blocks bitcoin@raspibolt:/mnt/hdd/bitcoin/
``
and also
```
rsync -r -e ssh /Users/noahtsutsui/Library/Application Support/Bitcoin/chainstate bitcoin@raspibolt:/mnt/hdd/bitcoin/
```
`-r` will recursively download subdirs

ACregan, 2018-06-23:
HI @Stadicus, Thanks for the great tutorial, I thought I'd make you aware of a problem I encountered, I didnt deviate from your instructions except I installed bitcoin v0.16.1 instead. I copied my up to date blockchain over from my desktop windows pc without problems and it seemed to be fine whilst running in testnet. However when I moved over to mainnet, updated my config and rebooted bitcoind wouldnt start (and would keep attempting to start in a loop) becuase it needed to "-reindex or -reindex-chainstate". I have done the latter but I think its effectively downloading the blockchain again, Its been going for about 14 hours now and its still only showing progress=0.105532 in the debug.log chatter. 
I've had a look around the docs to see if I've taken a wrong turn but it seems I've done everything correctly but for whatever reason, bitcoin cli 0.16.1 has rejected the copy of the blockchain I copied over from by windows pc. 
Any guidance you can offer would be great, once again: Thanks for a fantastic guide, a great contribution to the BTC/LND community, Cheers!

ACregan, 2018-06-24:
I can only conclude that my copy of the blockchain on my windows pc was generated without the txindex=1 flag. I reuploaded it and the same result, I'm just going to let it finish the -reindex-chainstate (ETA 4-5 days by my estimate). *sob*
There were a few things I should clarify:
I used filezilla and sftp, not winSCP as you recommended but I dont think that made any difference: I connected using filezilla and uploaded the blockchain as the 'bitcoin' user.
My windows blockchain was originally built with 0.14.0, I upgraded it to 0.16.1 on the windows pc and I had to wait a while whilst it updated 'UTXO table' or something like that, still had the same problem: bitcoin wouldnt start complaining that it needed re-index

ACregan, 2018-06-24:
yep, to test I just ran my windows client with the txindex=1 flag in the bitcoin.conf file and its reindexing the whole thing on the windows machine too. I guess I should have looked into this before messing with it.

Thanks for all your help and the excellent guide, Cheers!

Stadicus, 2018-06-24:
The txindex=1 is absolutely essential, as stated in the guide. 

You can also add it later on your computer and let that machine do the reindexing, that's faster than on the Pi.