raspibolt/raspibolt #94: Syncing the testnet

elkimek, 2018-06-21:
Hi, Stadicus,

thank you for the guide! May I ask how long it should it take syncing the testnet? I see there is a copying the mainnet blockchain in your guide, but how about the testnet? I never done it so I have no idea how long it will take on Raspi3 B+ with SSD, thanks!

nipponese, 2018-06-21:
I synced a copy using Bitcoin-qt on Mac, then copied the `blocks` and `chainstate` over on a USB thumb drive. Bitcoind still required me to run `bitcoind -reindex-chainstate` which took about 8 hours.

elkimek, 2018-06-21:
So I let it over night and I'm still at  "verificationprogress": 5.487615292986392e-08 
chain: says test, so it should be the mainnet, right?

nipponese, 2018-06-21:
@elkimek What does your `bitcoin.conf` say?

Stadicus, 2018-06-22:
@elkimek That's almost no progress. Testnet should be at 0.999 by now. As @nipponese mentioned: are you sure you are on testnet? Can you post the output of the following commands? 
* `sudo systemctl status bitcoind`
* `sudo tail /home/bitcoin/.bitcoin/debug.log -n 20`
* `bitcoin-cli -getinfo`
* `cat /home/bitcoin/.bitcoin/bitcoin.conf` (remove `rpcpassword` first)

Cees2daBees, 2018-06-22:
@elkimelk Sounds similar to an issue I had when I first set up my testnet, was able to solve it here: https://github.com/Stadicus/guides/issues/56
If you wanna give that a try.

elkimek, 2018-06-23:
Hey guys,

I went the way where I dl'ed and synced the testnet on stronger machine. Then I tried to SCP it to my external SSD plugged to RPI3B+ and somewhere around 75 % of copying the files I got DC from SSH and the device seems dead. LEDs are working, PI is heating up but I can't see the device in my router nor I can connect to it. So I think its bricked. Another piece on my way with heatsink tho :D

Stadicus, 2018-06-24:
Hm, never heard of bricking a Pi this way. And SCP does not really require heavy CPU resources. Did you try to restart by cutting power? Reflash the SD card (everything important is on the hard disk)?

elkimek, 2018-06-26:
Funny story here. 

The FS on SD card was just bricked, reflash helped. 
The I killed the Pi with my wedding ring lol. Don't get married 🗡 
I'm actually syncing the LN testnet right now on the new Pi :)

Anyway syncing the testnet seemed to slow anyway, so I just did that on my main PC and used MobaXterm (great piece of software btw) to SCP it on SSD, which also took bit long.

nipponese, 2018-06-27:
@elkimek Can I ask, did you disable swap on the card (and re-enable on the HDD) correctly?

elkimek, 2018-06-27:
I followed the guide exactly, so I guess I did. How do I verify?

nipponese, 2018-06-27:
@elkimek Maybe try `sudo find / -name swapfile`

nipponese, 2018-06-27:
@elkimek try `swapon -s`

elkimek, 2018-06-27:
``Filename                                Type            Size    Used    Priority
/var/swap                               file            1023996 0       -2
``

nipponese, 2018-06-27:
@elkimek Yeah, it looks like you're still using the SD card for your swap file. 
Re-do this section: [Moving the Swap File](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#moving-the-swap-file)

elkimek, 2018-06-27:
```Filename                                Type            Size    Used    Priority
/mnt/hdd/swapfile                       file            1023996 0       -2
```

Seems I left a hashtag in> sudo nano /etc/dphys-swapfile

Thanks! :)

nipponese, 2018-06-27:
@elkimek I don't really know too much about why a lot of file read/writes can degrade the SD card, but it may be permanently damaged.

elkimek, 2018-06-27:
Something definetly went south because whenever I launch bitcoind, it just goes off after few seconds. I'll try t reflash the card and will start over.

nipponese, 2018-06-27:
@elkimek What do you get when you run `sudo tail -f /home/bitcoin/.bitcoin/testnet3/debug.log`?

elkimek, 2018-06-27:
Is the 100% /dev/root alright?

df is

```admin@ElkimLND:~ $ df
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root        7410520  6986768     58192 100% /
devtmpfs          495484        0    495484   0% /dev
tmpfs             500092        0    500092   0% /dev/shm
tmpfs             500092    12908    487184   3% /run
tmpfs               5120        4      5116   1% /run/lock
tmpfs             500092        0    500092   0% /sys/fs/cgroup
/dev/mmcblk0p1     43436    22138     21299  51% /boot
/dev/sda1      479668424 15601412 439631428   4% /mnt/hdd
tmpfs             100016        0    100016   0% /run/user/1001
```

I have the swap correct this time

```
admin@ElkimLND:~ $ swapon -s
Filename                                Type            Size    Used    Priority
/mnt/hdd/swapfile                       file            1023996 0       -2
```

Because I'm having weird issues where bitcoind shuts down suddenly when /dev/root reaches 100%

nipponese, 2018-06-27:
Yeah, that's very strange, `/dev/root`. You should only have used about 3600000 kbytes on your card. Are you sure your symlinks pointing to your hard drive setup correctly?
What is the output of:
`ls -lart /home/bitcoin/.bitcoin`
`ls -l /home/bitcoin/.bitcoin/testnet3`

nipponese, 2018-06-27:
Also, did you delete `/var/swap`?

elkimek, 2018-06-27:
Both are located @ /mnt/hdd/

Problem is that syncing the testnet to 0.99 is filling the /dev/root

I do following:

Follow the guide to the last step of Bitcoin part.
Bitcoin-cli stop
SCP the blockchain and chainstate folder from my main PC to /mnt/hdd/bitcoin/testnet3
bitcoid when done

But during the verificationprogress the /dev/root gets full

nipponese, 2018-06-27:
It sounds like your symlinks aren't working.
Again, what is the output from running these commands?
`ls -l /home/bitcoin/.bitcoin
`ls -l /home/bitcoin/.bitcoin/testnet3`

Also what's the output from running?
`ls -l /var/`?

elkimek, 2018-06-27:
```
admin@ElkimLND:~ $ ls -l /home/bitcoin/.bitcoin ls -l /home/bitcoin/.bitcoin/testnet3
ls: cannot access 'ls': No such file or directory
lrwxrwxrwx 1 bitcoin bitcoin   16 Jun 27 22:08 /home/bitcoin/.bitcoin -> /mnt/hdd/bitcoin

ls: cannot open directory '/home/bitcoin/.bitcoin/testnet3': Permission denied
```
```
admin@ElkimLND:~ $ ls -l /var/
total 36
drwxr-xr-x  2 root root  4096 Jun 27 21:29 backups
drwxr-xr-x 12 root root  4096 Apr 18 03:24 cache
drwxr-xr-x 45 root root  4096 Jun 27 21:59 lib
drwxrwsr-x  2 root staff 4096 Mar 12 15:03 local
lrwxrwxrwx  1 root root     9 Apr 18 02:03 lock -> /run/lock
drwxr-xr-x  5 root root  4096 Jun 27 23:19 log
drwxrwsr-x  2 root mail  4096 Apr 18 02:03 mail
drwxr-xr-x  2 root root  4096 Apr 18 02:03 opt
lrwxrwxrwx  1 root root     4 Apr 18 02:03 run -> /run
drwxr-xr-x  5 root root  4096 Apr 18 02:41 spool
drwxrwxrwt  3 root root  4096 Jun 27 22:12 tmp
```

nipponese, 2018-06-27:
Can you run `ls -l /home/bitcoin/.bitcoin/testnet3` separately?

elkimek, 2018-06-27:
```
admin@ElkimLND:~ $ ls -l /home/bitcoin/.bitcoin/testnet3
ls: cannot open directory '/home/bitcoin/.bitcoin/testnet3': Permission denied
```

works from bitcoin user tho

```
bitcoin@ElkimLND:/home/admin $ ls -l /home/bitcoin/.bitcoin/testnet3
total 10944
-rw------- 1 bitcoin bitcoin       37 Jun 27 22:11 banlist.dat
drwx------ 3 bitcoin bitcoin     4096 Jun 27 22:37 blocks
drwx------ 2 bitcoin bitcoin    20480 Jun 27 22:41 chainstate
-rw------- 1 bitcoin bitcoin 10370264 Jun 28 00:17 debug.log
-rw------- 1 bitcoin bitcoin   247985 Jun 27 22:37 fee_estimates.dat
-rw------- 1 bitcoin bitcoin       17 Jun 27 22:37 mempool.dat
-rw------- 1 bitcoin bitcoin   543172 Jun 27 22:37 peers.dat
drwx------ 2 bitcoin bitcoin     4096 Jun 27 22:11 wallets
```

nipponese, 2018-06-27:
Try this taken from [stack overflow](https://askubuntu.com/questions/266825/what-do-i-do-when-my-root-filesystem-is-full)
`sudo du -hsx /* | sort -rh | head -n 40`

elkimek, 2018-06-28:
Solved. I SCPed the testnet3 folder before first launching the bitcoind. The "server" was restarting because I had another LND started when I had ZAP wallet in the background ^^

Stadicus, 2018-06-29:
So everything working fine? 

Just as an additional input: I once had an issue with mounting the hard disk. The symlink then still pointed to the SD card (the real directory `/mnt/hdd/bitcoin`) and filled the card. But when the hard disk was mounted, it was mounted on top and this real directory was no longer visible, while the SD card was still full. This could be the /dev/root on 100%. Took me quite some time to figure that out.

Solution was to unmount the hard disk, check if there was a real directory (`/mnt/hdd/bitcoin`) and delete that, and then mount the hard disk again.

elkimek, 2018-06-30:
I'm having the same issue right now with mainnet. I SCPed the chainstate, blocks successfully, bitcoind accepts the copied files, started bitcoind with no error and during the initial verification process the /dev/root gets filled and bitcoind crash when the SD card gets filled on 100 %. 

@Stadicus can you please elaborate the solution? Will I need to copy the blocks again? It takes 4,5hrs even on SSD.

Stadicus, 2018-07-07:
Sorry for the late reply, just came back from Building on Bitcoin in Lisbon. Didn't even take a computer with me... :-)

What I was trying to describe was an issue with automatically mounting the hard disk. The symlink `/home/bitcoin/.bitcoin` point to `/mnt/hdd/bitcoin/` (representing the external hdd) and all application data is stored in subdirectories. 

If the external hdd is not mounted, chances are that these subdirectories are created on the sd card. All data, and especially the blockchain, is then stored on the sdcard.

What gave me a headache was that the sdcard was almost full, but then the external hdd was mounted correctly with the **mount point** `/mnt/hdd/bitcoin/`, over the **directory** `/mnt/hdd/bitcoin/`. All the previous data was still on the sdcard, but no longer visible using `du` until the external hdd was manually unmounted. 

Not sure if this is the case, but it sure seems like the application data is stored on the sdcard, thus 
* either the symlink points to the sdcard, or
* the external hdd is not correctly mounted

elkimek, 2018-07-13:
Yup all good riding on mainnet for for two weeks now :)

closed 2018-07-13 by elkimek