raspibolt/raspibolt #95: Send Payment Issue

BitcoinVegan, 2018-06-25:
Hello @Stadicus Im trying to send payment invoice to @moabbit between two nodes, connection is true on lightning testnet.

Invoice
```lncli payinvoice 0315249fdd6717953599420c883b3ed1b6582b3335bccbfc9651baf1bbefdf4bce 1200 bolt11
[lncli] rpc error: code = Unknown desc = failed converting data to bytes: invalid character not part of charset: 98```

Error:

 ```lncli sendpayment 0315249fdd6717953599420c883b3ed1b6582b3335bccbfc9651baf1bbefdf4bce 1200 bolt11
[lncli] encoding/hex: invalid byte: U+006F 'o'```

Any suggestions?

Moabbit, 2018-06-25:
@stadicus it's lncli add invoice and the other lncli pay invoice

BitcoinVegan, 2018-06-26:
I've used that command but I'm not understanding if or even what the payment hash is

BitcoinVegan, 2018-06-26:
@Stadicus this is an invoice I tried to send to another person, using the same command

` lncli payinvoice 0315249fdd6717953599420c883b3ed1b6582b3335bccbfc9651baf1bbefdf4bce 1200 bolt11lncli payinvoice 0315249fdd6717953599420c883b3ed1b6582b3335bccbfc9651baf1bbefdf4bce 1200 bolt11
[lncli] rpc error: code = Unknown desc = failed converting data to bytes: invalid character not part of charset: 98`

Stadicus, 2018-06-26:
Well, this does not look like a valid payment request, maybe you try to use the r_hash? I just created one:

```
lncli addinvoice 1200
{
        "r_hash": "a68c3f526e2f5e2733c1ddef97db90162ee9595790598274d36e506bcbbd32db",
        "pay_req": "lnbc12u1pdn9vxxpp556xr75nw9a0zwv7pmhhe0kuszchwjk2hjpvcyaxndegxhjaaxtdsdqqcqzys90nk3srv2qgaklxf6cghxk2x8yfdkqwjsm7a3ucx3g59rxk6cg4r9yzcc57a58mrg9nujr5kdcz00jav4hl73zkdj8vmmfg8zzv2l8qqt4x7x4"
}
```
You would pay by using the pay_req value with 
```
lncli payinvoice lnbc12u1pdn9vxxpp556xr75nw9a0zwv7pmhhe0kuszchwjk2hjpvcyaxndegxhjaaxtdsdqqcqzys90nk3srv2qgaklxf6cghxk2x8yfdkqwjsm7a3ucx3g59rxk6cg4r9yzcc57a58mrg9nujr5kdcz00jav4hl73zkdj8vmmfg8zzv2l8qqt4x7x4
```

BitcoinVegan, 2018-06-28:
i received this after entering.

` lncli payinvoice lnbc12u1pdn9vxxpp556xr75nw9a0zwv7pmhhe0kuszchwjk2hjpvcyaxndegxhjaaxtdsdqqcqzys90nk3srv2qgaklxf6cghxk2x8yfdkqwjsm7a3ucx3g59rxk6cg4r9yzcc57a58mrg9nujr5kdcz00jav4hl73zkdj8vmmfg8zzv2l8qqt4x7x4
[lncli] rpc error: code = Unknown desc = unknown network`

BitcoinVegan, 2018-06-28:
my apologies. I created an invoice and got this.

`lncli addinvoice 1200
{
        "r_hash": "024055feafc876e639a7ba45c9c97f42b5ed1412b86c40027946356713e3becd",
        "pay_req": "lntb12u1pdnfm9fpp5qfq9tl40epmwvwd8hfzunjtlg26769qjhpkyqqnegc6kwylrhmxsdqqcqzyst8dqzgrz3udh056x2rc7ttehq7al5v04vzallw8mxqayslvwesw3yn9slvjhs7z5c856wuw8nhhk88x9ypj2c4j584yx5tts9t37upqpqfzhdr"`

Stadicus, 2018-06-29:
Oh, my invoice was for mainnet, and you are on testnet, sorry.

The invoice looks good. Someone else should be able to pay it. You can generate remote invoces by "buying" stuff, eg. https://starblocks.acinq.co or https://www.yalls.org and then "pay" them with `lncli payinvoice [loooooong invoice text]`

Stadicus, 2018-06-29:
So up until now I don't think it's an issue with LND or the RaspiBolt guide, just usage questions. Try reading the mainnet section (long list of commands), the FAQ and other resources like https://dev.lightning.community to educate yourself. And feel free to write about your own learnings somewhere, eg. on Medium.

closed 2018-06-29 by Stadicus