raspibolt/raspibolt #97: Connection issues with LND?

elkimek, 2018-06-27:
Hey,

I have running LND to the point where I got coins from faucet and they are successfully confirmed.

However I don't have any peers nor channels opened and getting following errors when 

$ sudo journalctl -f -u lnd

```
-- Logs begin at Thu 2016-11-03 18:16:42 CET. --
Jun 27 12:35:24 ElkimLND lnd[8446]: 2018-06-27 12:35:24.570 [INF] DISC: Attempting to bootstrap with: BOLT-0010 DNS Seed: [[test.nodes.lightning.directory soa.nodes.lightning.directory]]
Jun 27 12:35:46 ElkimLND lnd[8446]: 2018-06-27 12:35:46.511 [INF] CRTR: Pruning channel graph using block 0000000000007173892e83a178da8ea9bb605105a629cc9af3838d235826c20f (height=1326481)
Jun 27 12:35:46 ElkimLND lnd[8446]: 2018-06-27 12:35:46.517 [INF] CRTR: Block 0000000000007173892e83a178da8ea9bb605105a629cc9af3838d235826c20f (height=1326481) closed 0 channels
Jun 27 12:35:46 ElkimLND lnd[8446]: 2018-06-27 12:35:46.556 [INF] NTFN: New block: height=1326481, sha=0000000000007173892e83a178da8ea9bb605105a629cc9af3838d235826c20f
Jun 27 12:35:46 ElkimLND lnd[8446]: 2018-06-27 12:35:46.558 [INF] UTXN: Attempting to graduate height=1326481: num_kids=0, num_babies=0
Jun 27 12:35:57 ElkimLND lnd[8446]: 2018-06-27 12:35:57.317 [INF] RPCS: [listchannels] fetched 0 channels from DB
Jun 27 12:35:59 ElkimLND lnd[8446]: 2018-06-27 12:35:59.586 [ERR] DISC: Unable to query bootstrapper BOLT-0010 DNS Seed: [[test.nodes.lightning.directory soa.nodes.lightning.directory]]: lookup soa.nodes.lightning.directory on 1.0.0.1:53: read udp 10.0.1.9:50074->1.0.0.1:53: i/o timeout
Jun 27 12:35:59 ElkimLND lnd[8446]: 2018-06-27 12:35:59.588 [INF] DISC: Obtained 0 addrs to bootstrap network with
Jun 27 12:35:59 ElkimLND lnd[8446]: 2018-06-27 12:35:59.589 [INF] DISC: Attempting to bootstrap with: Authenticated Channel Graph
Jun 27 12:35:59 ElkimLND lnd[8446]: 2018-06-27 12:35:59.644 [INF] DISC: Attempting to bootstrap with: BOLT-0010 DNS Seed: [[test.nodes.lightning.directory soa.nodes.lightning.directory]]
```
Router setting should be fine:

[image]

and

[image]

Any advice? Thank you.

elkimek, 2018-06-27:
I tried sudo reboot and after:

$ lncli unlock
```lnd successfully unlocked!```

$ sudo journalctl -f -u lnd
```
-- Logs begin at Thu 2016-11-03 18:16:42 CET. --
Jun 27 13:06:28 ElkimLND lnd[2014]: 2018-06-27 13:06:28.481 [INF] CHDB: Checking for schema update: latest_version=0, db_version=0
Jun 27 13:06:28 ElkimLND lnd[2014]: 2018-06-27 13:06:28.491 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
Jun 27 13:06:28 ElkimLND lnd[2014]: 2018-06-27 13:06:28.492 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Jun 27 13:06:28 ElkimLND lnd[2014]: 2018-06-27 13:06:28.493 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create wallet, or `lncli unlock` to unlock already created wallet.
Jun 27 13:07:12 ElkimLND lnd[2014]: 2018-06-27 13:07:12.550 [INF] LNWL: Opened wallet
Jun 27 13:07:12 ElkimLND lnd[2014]: 2018-06-27 13:07:12.905 [INF] LTND: Primary chain is set to: bitcoin
Jun 27 13:07:12 ElkimLND lnd[2014]: 2018-06-27 13:07:12.905 [INF] LTND: Initializing bitcoind backed fee estimator
Jun 27 13:07:12 ElkimLND systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Jun 27 13:07:12 ElkimLND systemd[1]: lnd.service: Unit entered failed state.
Jun 27 13:07:12 ElkimLND systemd[1]: lnd.service: Failed with result 'exit-code'.
Jun 27 13:08:13 ElkimLND systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Jun 27 13:08:13 ElkimLND systemd[1]: Stopped LND Lightning Daemon.
Jun 27 13:08:30 ElkimLND systemd[1]: Started LND Lightning Daemon.
Jun 27 13:08:30 ElkimLND lnd[2810]: Attempting automatic RPC configuration to bitcoind
Jun 27 13:08:30 ElkimLND lnd[2810]: Automatically obtained bitcoind's RPC credentials
Jun 27 13:08:30 ElkimLND lnd[2810]: 2018-06-27 13:08:30.453 [INF] LTND: Version 0.4.2-beta commit=7cf5ebe2650b6798182e10be198c7ffc1f1d6e19
Jun 27 13:08:30 ElkimLND lnd[2810]: 2018-06-27 13:08:30.453 [INF] LTND: Active chain: Bitcoin (network=testnet)
Jun 27 13:08:30 ElkimLND lnd[2810]: 2018-06-27 13:08:30.454 [INF] CHDB: Checking for schema update: latest_version=0, db_version=0
Jun 27 13:08:30 ElkimLND lnd[2810]: 2018-06-27 13:08:30.464 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
Jun 27 13:08:30 ElkimLND lnd[2810]: 2018-06-27 13:08:30.464 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Jun 27 13:08:30 ElkimLND lnd[2810]: 2018-06-27 13:08:30.464 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create wallet, or `lncli unlock` to unlock already created wallet.
```

elkimek, 2018-06-27:
Hm, It seems my testnet bloackchain is resyncing again :( had 0.999999 yesterday and now I have again  5.473495874562393e-08.

nipponese, 2018-06-27:
What happens when you run `curl <your public ip>:18333` If you get `curl: (52) Empty reply from server` then you know at least your testnet port is open correct. But if it hangs, then there is something wrong with your port forwarding.

elkimek, 2018-06-27:
I went with reflashing the card and started over. Will get back once I do all the setup and SCP the testnet :)

elkimek, 2018-06-28:
Getting the ```curl: (52) Empty reply from server```

but still seeing 

```
Jun 28 14:16:46 ElkimLND lnd[3877]: 2018-06-28 14:16:46.843 [ERR] DISC: Unable to query bootstrapper BOLT-0010 DNS Seed: [[test.nodes.lightning.directory soa.nodes.lightning.directory]]: lookup soa.nodes.lightning.directory on 1.0.0.1:53: read udp 10.0.1.9:34127->1.0.0.1:53: i/o timeout

``` 
I don't get it where the 1.0.0.1:53 cames from tho

and have no peers no opened channel

elkimek, 2018-06-28:
Ok, didn't helped, here is the larger part of the journal 

```
Jun 28 14:58:10 ElkimLND lnd[1285]: 2018-06-28 14:58:10.585 [INF] CRTR: Block 000000000000f3db7af1d3a251756eea4a1917ee0c7453436f47f20f18fb1659 (height=1342815) closed 0 channels
Jun 28 14:58:11 ElkimLND lnd[1285]: 2018-06-28 14:58:11.999 [ERR] CMGR: Can't accept connection: chacha20poly1305: message authentication failed
Jun 28 14:58:18 ElkimLND lnd[1285]: 2018-06-28 14:58:18.753 [INF] CRTR: Pruning channel graph using block 00000000000028ec28b3062620b080ec400f8e1189a6c1e430276375476a6c41 (height=1342816)
Jun 28 14:58:18 ElkimLND lnd[1285]: 2018-06-28 14:58:18.763 [INF] NTFN: New block: height=1342816, sha=00000000000028ec28b3062620b080ec400f8e1189a6c1e430276375476a6c41
Jun 28 14:58:18 ElkimLND lnd[1285]: 2018-06-28 14:58:18.763 [INF] CRTR: Block 00000000000028ec28b3062620b080ec400f8e1189a6c1e430276375476a6c41 (height=1342816) closed 0 channels
Jun 28 14:58:18 ElkimLND lnd[1285]: 2018-06-28 14:58:18.764 [INF] UTXN: Attempting to graduate height=1342816: num_kids=0, num_babies=0
Jun 28 14:58:20 ElkimLND lnd[1285]: 2018-06-28 14:58:20.255 [INF] CRTR: Pruning channel graph using block 000000000000656572dc4ee1928661790e5c5b603ba099a01a079eb397f92394 (height=1342817)
Jun 28 14:58:20 ElkimLND lnd[1285]: 2018-06-28 14:58:20.273 [INF] CRTR: Block 000000000000656572dc4ee1928661790e5c5b603ba099a01a079eb397f92394 (height=1342817) closed 0 channels
Jun 28 14:58:20 ElkimLND lnd[1285]: 2018-06-28 14:58:20.285 [INF] NTFN: New block: height=1342817, sha=000000000000656572dc4ee1928661790e5c5b603ba099a01a079eb397f92394
Jun 28 14:58:20 ElkimLND lnd[1285]: 2018-06-28 14:58:20.286 [INF] UTXN: Attempting to graduate height=1342817: num_kids=0, num_babies=0
Jun 28 14:58:25 ElkimLND lnd[1285]: 2018-06-28 14:58:25.870 [INF] CRTR: Pruning channel graph using block 00000000000054557e369e7388463defeff8d7a0bc44324d64fb1934757b3ddf (height=1342818)
Jun 28 14:58:25 ElkimLND lnd[1285]: 2018-06-28 14:58:25.879 [INF] NTFN: New block: height=1342818, sha=00000000000054557e369e7388463defeff8d7a0bc44324d64fb1934757b3ddf
Jun 28 14:58:25 ElkimLND lnd[1285]: 2018-06-28 14:58:25.880 [INF] UTXN: Attempting to graduate height=1342818: num_kids=0, num_babies=0
Jun 28 14:58:25 ElkimLND lnd[1285]: 2018-06-28 14:58:25.883 [INF] CRTR: Block 00000000000054557e369e7388463defeff8d7a0bc44324d64fb1934757b3ddf (height=1342818) closed 0 channels
Jun 28 14:58:26 ElkimLND lnd[1285]: 2018-06-28 14:58:26.280 [ERR] CMGR: Can't accept connection: chacha20poly1305: message authentication failed
Jun 28 14:58:30 ElkimLND lnd[1285]: 2018-06-28 14:58:30.455 [ERR] DISC: Unable to query bootstrapper BOLT-0010 DNS Seed: [[test.nodes.lightning.directory soa.nodes.lightning.directory]]: lookup soa.nodes.lightning.directory on 1.0.0.1:53: read udp 10.0.1.9:39957->1.0.0.1:53: i/o timeout
Jun 28 14:58:30 ElkimLND lnd[1285]: 2018-06-28 14:58:30.455 [INF] DISC: Obtained 0 addrs to bootstrap network with
Jun 28 14:58:30 ElkimLND lnd[1285]: 2018-06-28 14:58:30.455 [INF] DISC: Attempting to bootstrap with: Authenticated Channel Graph
Jun 28 14:58:30 ElkimLND lnd[1285]: 2018-06-28 14:58:30.512 [INF] DISC: Attempting to bootstrap with: BOLT-0010 DNS Seed: [[test.nodes.lightning.directory soa.nodes.lightning.directory]]
Jun 28 14:58:32 ElkimLND lnd[1285]: 2018-06-28 14:58:32.635 [INF] CRTR: Pruning channel graph using block 0000000000008179a07c9d3982a1f26cb57134b562cda134778357b94d0384ec (height=1342819)
Jun 28 14:58:32 ElkimLND lnd[1285]: 2018-06-28 14:58:32.643 [INF] CRTR: Block 0000000000008179a07c9d3982a1f26cb57134b562cda134778357b94d0384ec (height=1342819) closed 0 channels
Jun 28 14:58:32 ElkimLND lnd[1285]: 2018-06-28 14:58:32.651 [INF] NTFN: New block: height=1342819, sha=0000000000008179a07c9d3982a1f26cb57134b562cda134778357b94d0384ec
Jun 28 14:58:32 ElkimLND lnd[1285]: 2018-06-28 14:58:32.652 [INF] UTXN: Attempting to graduate height=1342819: num_kids=0, num_babies=0
Jun 28 14:58:38 ElkimLND lnd[1285]: 2018-06-28 14:58:38.456 [INF] CRTR: Pruning channel graph using block 0000000000004679586bf1d660a016671ee16dad271b2e2dbce7f22204138ea5 (height=1342820)
Jun 28 14:58:38 ElkimLND lnd[1285]: 2018-06-28 14:58:38.474 [INF] NTFN: New block: height=1342820, sha=0000000000004679586bf1d660a016671ee16dad271b2e2dbce7f22204138ea5
Jun 28 14:58:38 ElkimLND lnd[1285]: 2018-06-28 14:58:38.476 [INF] UTXN: Attempting to graduate height=1342820: num_kids=0, num_babies=0
Jun 28 14:58:38 ElkimLND lnd[1285]: 2018-06-28 14:58:38.478 [INF] CRTR: Block 0000000000004679586bf1d660a016671ee16dad271b2e2dbce7f22204138ea5 (height=1342820) closed 0 channels
Jun 28 14:58:57 ElkimLND lnd[1285]: 2018-06-28 14:58:57.616 [INF] CRTR: Pruning channel graph using block 00000000000012024ce5b167a5a6bdd486a75bfc8cfef4d1767b0df4f889475f (height=1342821)
Jun 28 14:58:57 ElkimLND lnd[1285]: 2018-06-28 14:58:57.628 [INF] CRTR: Block 00000000000012024ce5b167a5a6bdd486a75bfc8cfef4d1767b0df4f889475f (height=1342821) closed 0 channels
Jun 28 14:58:57 ElkimLND lnd[1285]: 2018-06-28 14:58:57.633 [INF] NTFN: New block: height=1342821, sha=00000000000012024ce5b167a5a6bdd486a75bfc8cfef4d1767b0df4f889475f
Jun 28 14:58:57 ElkimLND lnd[1285]: 2018-06-28 14:58:57.633 [INF] UTXN: Attempting to graduate height=1342821: num_kids=0, num_babies=0
Jun 28 14:58:59 ElkimLND lnd[1285]: 2018-06-28 14:58:59.970 [INF] NTFN: Block disconnected from main chain: height=1342821, sha=00000000000012024ce5b167a5a6bdd486a75bfc8cfef4d1767b0df4f889475f
[... cut ...]
Jun 28 14:58:59 ElkimLND lnd[1285]: 2018-06-28 14:58:59.990 [INF] CRTR: Pruning channel graph using block 00000000000031d1cc5c41d085059ad83a5cd7bd5949bc3659843a09717c607c (height=1342822)
Jun 28 14:58:59 ElkimLND lnd[1285]: 2018-06-28 14:58:59.991 [INF] NTFN: New block: height=1342822, sha=00000000000031d1cc5c41d085059ad83a5cd7bd5949bc3659843a09717c607c
Jun 28 14:58:59 ElkimLND lnd[1285]: 2018-06-28 14:58:59.997 [INF] UTXN: Attempting to graduate height=1342822: num_kids=0, num_babies=0
Jun 28 14:59:00 ElkimLND lnd[1285]: 2018-06-28 14:59:00.004 [INF] CRTR: Block 00000000000031d1cc5c41d085059ad83a5cd7bd5949bc3659843a09717c607c (height=1342822) closed 0 channels
```

nipponese, 2018-06-28:
Sorry, I should have specified to try using curl from another machine, not the Raspi.

I am curious, what happens you try `lncli connect` to something likes yalls? `02212d3ec887188b284dbb7b2e6eb40629a6e14fb049673f22d2a0aa05f902090e@testnet-lnd.yalls.org`?

elkimek, 2018-06-28:
```
admin@ElkimLND:~ $ lncli connect 02212d3ec887188b284dbb7b2e6eb40629a6e14fb049673f22d2a0aa05f902090e@testnet-lnd.yalls.org
{

}
admin@ElkimLND:~ $ sudo journalctl -f -u lnd
-- Logs begin at Thu 2016-11-03 18:16:42 CET. --
Jun 28 16:52:05 ElkimLND lnd[1285]: 2018-06-28 16:52:05.350 [INF] DISC: Attempting to bootstrap with: Authenticated Channel Graph
Jun 28 16:52:05 ElkimLND lnd[1285]: 2018-06-28 16:52:05.379 [INF] CRTR: New channel discovered! Link connects 025b179fc069788b475b60d362fdde796cbf729fb9f3d95090ef37c079fa7eaa86 and 03933884aaf1d6b108397e5efe5c86bcf2d8ca8d2f700eda99db9214fc2712b134 with ChannelPoint(3e3bff16d362c5f38adb37a41e06fc11c35854770a20466261d55cb347707174:0): chan_id=1417383737910427648, capacity=0.01 BTC
Jun 28 16:52:05 ElkimLND lnd[1285]: 2018-06-28 16:52:05.413 [INF] DISC: Obtained 4 addrs to bootstrap network with
Jun 28 16:52:05 ElkimLND lnd[1285]: 2018-06-28 16:52:05.466 [INF] NTFN: New block: height=1343936, sha=000000000000fa418cc974de17c80d1c2db2d0256fa7238206bf2fb3a9d8e822
Jun 28 16:52:05 ElkimLND lnd[1285]: 2018-06-28 16:52:05.467 [INF] UTXN: Attempting to graduate height=1343936: num_kids=0, num_babies=0
Jun 28 16:52:05 ElkimLND lnd[1285]: 2018-06-28 16:52:05.537 [INF] CRTR: New channel discovered! Link connects 03b1cf5623ca6757d49de3b6e2b9340065ba991c75b8e9cd8aec51dc54322cbd1d and 03fd0aebde8713e9c311f22468d3d0524e788b1ef57f4cda41bf5b5a2300fc5cd6 with ChannelPoint(9d25d46a160e911ea987e8b588a5823d44087d5c6dd62d637e7d70c4f004d604:0): chan_id=1417383737910689792, capacity=0.16777216 BTC
Jun 28 16:52:05 ElkimLND lnd[1285]: 2018-06-28 16:52:05.559 [INF] CRTR: Pruning channel graph using block 000000000000fa418cc974de17c80d1c2db2d0256fa7238206bf2fb3a9d8e822 (height=1343936)
Jun 28 16:52:05 ElkimLND lnd[1285]: 2018-06-28 16:52:05.567 [INF] CRTR: Block 000000000000fa418cc974de17c80d1c2db2d0256fa7238206bf2fb3a9d8e822 (height=1343936) closed 0 channels
Jun 28 16:52:05 ElkimLND lnd[1285]: 2018-06-28 16:52:05.655 [ERR] SRVR: unable to connect to 020a3ce6e6893749bbcdb67ac67570e816a17c678bbcb6b12b0325f3fec036a014@189.4.126.1:9735: dial tcp 189.4.126.1:9735: connect: connection refused
Jun 28 16:52:05 ElkimLND lnd[1285]: 2018-06-28 16:52:05.823 [INF] CRTR: New channel discovered! Link connects 03023d5233b61281abd5187f27c9ce422183b3374c38ee6c9ec8cf4221e5a92076 and 03933884aaf1d6b108397e5efe5c86bcf2d8ca8d2f700eda99db9214fc2712b134 with ChannelPoint(d89e211f4a021be102445b56ee6acf82ddcce1dd5776748f6165555224bbbfd3:0): chan_id=1417384837415501824, capacity=0.0005 BTC
Jun 28 16:52:05 ElkimLND lnd[1285]: 2018-06-28 16:52:05.994 [INF] CRTR: Updated vertex data for node=025b179fc069788b475b60d362fdde796cbf729fb9f3d95090ef37c079fa7eaa86
Jun 28 16:52:06 ElkimLND lnd[1285]: 2018-06-28 16:52:06.075 [INF] CRTR: New channel discovered! Link connects 03023d5233b61281abd5187f27c9ce422183b3374c38ee6c9ec8cf4221e5a92076 and 03933884aaf1d6b108397e5efe5c86bcf2d8ca8d2f700eda99db9214fc2712b134 with ChannelPoint(b2ab9f1d0d59e9ca459fc641019eaa30bf7c07307c8a8bd286be7e12953dac64:0): chan_id=1417387036442558464, capacity=0.0002 BTC
Jun 28 16:52:06 ElkimLND lnd[1285]: 2018-06-28 16:52:06.187 [INF] CRTR: Updated vertex data for node=03fd0aebde8713e9c311f22468d3d0524e788b1ef57f4cda41bf5b5a2300fc5cd6
Jun 28 16:52:06 ElkimLND lnd[1285]: 2018-06-28 16:52:06.197 [INF] CRTR: Updated vertex data for node=03fd0aebde8713e9c311f22468d3d0524e788b1ef57f4cda41bf5b5a2300fc5cd6
Jun 28 16:52:06 ElkimLND lnd[1285]: 2018-06-28 16:52:06.202 [INF] CRTR: Updated vertex data for node=03fd0aebde8713e9c311f22468d3d0524e788b1ef57f4cda41bf5b5a2300fc5cd6
Jun 28 16:52:06 ElkimLND lnd[1285]: 2018-06-28 16:52:06.256 [INF] CRTR: New channel discovered! Link connects 02e2deb195cd4de5f5b079ece20e5e20f0cceee720c5fd2b5410c9130cf25b5561 and 030f0bf260acdbd3edcad84d7588ec7c5df4711e87e6a23016f989b8d3a4147230 with ChannelPoint(b281ba1bb44738b0f750c55f3030c2401e437b10d16e63b99b3b0469b4ac18bb:0): chan_id=1417387036444393472, capacity=0.01 BTC
Jun 28 16:52:06 ElkimLND lnd[1285]: 2018-06-28 16:52:06.265 [INF] CRTR: New channel discovered! Link connects 0269a94e8b32c005e4336bfb743c08a6e9beb13d940d57c479d95c8e687ccbdb9f and 02e2deb195cd4de5f5b079ece20e5e20f0cceee720c5fd2b5410c9130cf25b5561 with ChannelPoint(513eca90f93c9c22ec60a2d93c5526aa2a088ce5e24438d641bae0e76ef31ad0:0): chan_id=1417387036444524544, capacity=0.01 BTC
Jun 28 16:52:06 ElkimLND lnd[1285]: 2018-06-28 16:52:06.576 [INF] CRTR: Updated vertex data for node=03023d5233b61281abd5187f27c9ce422183b3374c38ee6c9ec8cf4221e5a92076
Jun 28 16:52:06 ElkimLND lnd[1285]: 2018-06-28 16:52:06.633 [INF] CRTR: New channel discovered! Link connects 023ea0a53af875580899da0ab0a21455d9c19160c4ea1b7774c9d4be6810b02d2c and 03fd0aebde8713e9c311f22468d3d0524e788b1ef57f4cda41bf5b5a2300fc5cd6 with ChannelPoint(dd60d8fa92640e0e46278808600fc68142da3820c4bf7d63e29bd9e8f1141452:0): chan_id=1417390334977638400, capacity=0.16777216 BTC
Jun 28 16:52:06 ElkimLND lnd[1285]: 2018-06-28 16:52:06.663 [INF] CRTR: New channel discovered! Link connects 0269a94e8b32c005e4336bfb743c08a6e9beb13d940d57c479d95c8e687ccbdb9f and 03fd0aebde8713e9c311f22468d3d0524e788b1ef57f4cda41bf5b5a2300fc5cd6 with ChannelPoint(02a1c10b579a602c2ad589ead7eb875590a870b971bd7be1fbb1c73d1188a8f0:0): chan_id=1417390334977900544, capacity=0.16777216 BTC
Jun 28 16:52:06 ElkimLND lnd[1285]: 2018-06-28 16:52:06.889 [INF] CRTR: New channel discovered! Link connects 036b96e4713c5f84dcb8030592e1bd42a2d9a43d91fa2e535b9bfd05f2c5def9b9 and 03fd0aebde8713e9c311f22468d3 [... cut ...]
            "inbound": false,
            "ping_time": "0"
        },
        {
            "pub_key": "028fd42a3eb383bfb803e49f393fb11786108d879fcd46cdfa5ee283e3b896077b",
            "address": "45.33.59.121:9735",
            "bytes_sent": "613023",
            "bytes_recv": "601811",
            "sat_sent": "0",
            "sat_recv": "0",
            "inbound": false,
            "ping_time": "0"
        }
    ]
}
admin@ElkimLND:~ $ lncli listchannels
{
    "channels": [
    ]
}
admin@ElkimLND:~ $
```

nipponese, 2018-06-28:
Looks ok to me.

elkimek, 2018-06-28:
Ok now I guess I should wait until autopilot establish some channel, right?

nipponese, 2018-06-28:
Have you tried making any payment requests?

elkimek, 2018-06-28:
Solved, just took time to establish the channel :)