raspibolt/raspibolt #101: REQUEST: add the procedures to install eclair and c-lightning

Amiga500, 2018-06-29:
Hi @Stadicus,

why not add the procedures to install eclair and c-lightning as finally are both in beta?

So we can switch through the three implementation.. and see which one is the better for our use?

This weekend i think to start to try to add c-lightning.. and if i'll have nice news.. i'll add here all infos

johneakin, 2018-06-30:
Please do
I am keen to install clightning
Can this run alongside LND?

Stadicus, 2018-07-02:
Hej @Amiga500, good idea. I don't want to complicate the basic setup (seeing the current amount of issues and questions), but this could be a valuable extension to the Bonus section. Want to create a Pull Request?

johneakin, 2018-07-02:
This would be awesome
Especially that lightningcharge (https://github.com/ElementsProject/lightning-charge) is integrated with c-lightning

i.e. if you want to create a website that displays qr code invoices, that connects to your pi REMOTELY.
I don't think this is easy with LND, most documentation about the LND REST is for localhost?

johneakin, 2018-07-02:
https://medium.com/@meeDamian/c-lightning-node-on-rbp3-b950660fb835

johneakin, 2018-07-02:
I just set this up as per the above link, and appears to be working
Longest part is compiling the code (15mins)

I left out the Bitcoin RPC part, as we can re-use the one setup in this guide
Rest is all easy and doesnt conflict with any of the names from the LND instructions.

Happy to help if you get stuck

Stadicus, 2018-08-03:
After some pondering I think as this is a beginner's guide, it does not need to reflect every possibility and running multiple Lightning variants is not of real value for the average user. As I am working more on simplifying the overall solution, I will close this topic.

Please feel free to do your own additional guide, I can link to it form the bonus page.

closed 2018-08-03 by Stadicus