raspibolt/raspibolt #98: ssh connection closes immediately after password is entered

briansteven, 2018-06-29:
Hi, thanks for this awesome guide!  I have a problem where I get:

Connection to thundroid closed.

Immediately after putting in my ssh password in (and it's correct).  I followed everything in the guide exactly, and everything went smoothly, until I got to setting up the SSH keys.  I wanted to just leave it alone with password login for now.  Did that mess it up?  How can I un-mess it up?

Thanks!

Stadicus, 2018-06-29:
Did you configure anything regarding SSH Keys? Like in the file `/etc/ssh/sshd_config` or in your SSH client (eg. to use a public key)? It's no problem to just skip the whole SSH Key part, but you need to skip it completely.
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#login-with-ssh-keys

Could it have something to do with unsuccessful login attempts that are now blocked by fail2ban?

briansteven, 2018-06-29:
No I didn't touch any of the SSH key stuff.  If I run ssh with -v 'ssh -v root@thundroid', I get this (bottom chunk of the output):

`debug1: Authentications that can continue: publickey,password
debug1: Next authentication method: publickey
debug1: Trying private key: /home/brian/.ssh/id_rsa
debug1: Trying private key: /home/brian/.ssh/id_dsa
debug1: Trying private key: /home/brian/.ssh/id_ecdsa
debug1: Trying private key: /home/brian/.ssh/id_ed25519
debug1: Next authentication method: password
root@thundroid's password:
debug1: Authentication succeeded (password).
Authenticated to thundroid ([192.168.86.30]:22).
debug1: channel 0: new [client-session]
debug1: Requesting no-more-sessions@openssh.com
debug1: Entering interactive session.
debug1: pledge: network
debug1: client_input_global_request: rtype hostkeys-00@openssh.com want_reply 0
debug1: Sending environment.
debug1: Sending env LANG = C.UTF-8
Last login: Thu Jun 28 22:35:47 2018 from 192.168.86.29
debug1: client_input_channel_req: channel 0 rtype exit-status reply 0
debug1: client_input_channel_req: channel 0 rtype eow@openssh.com reply 0
debug1: channel 0: free: client-session, nchannels 1
Connection to thundroid closed.
Transferred: sent 2176, received 2600 bytes, in 0.0 seconds
Bytes per second: sent 172386.0, received 205975.9
debug1: Exit status 254`

This does show that the authentication succeeded, but then it just quits. hmmm.

briansteven, 2018-06-30:
I mounted my sdcard in windows and undid the edits for the limits (https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#increase-your-open-files-limit).  Now I'm able to login again.

Stadicus, 2018-07-02:
I am really not sure what causes this. So for now, everything works fine? Will close for now, can reopen if problem persists.

closed 2018-07-02 by Stadicus