raspibolt/raspibolt #103: Not able to find my lightning node on explorer.acinq.co

johneakin, 2018-06-30:
My node is up and running
And automatically connected to some channels
However I cant find it listed on explorer.acinq.co

Any idea why that is?

Stadicus, 2018-07-02:
The LN explorers are also just a node that makes its own view of the network public. This may not be accurate due to various reasons and should not be expected to be 100% true. Sometimes it just takes some time.

Are you on Testnet? Did you try https://1ml.com/testnet/ ?

johneakin, 2018-07-02:
Yes I am on testnet
and I can see my node listed on that link you sent.
So we think this is a fault with explorer.acinq.co?

Stadicus, 2018-07-02:
Yes, definitely. Don't worry about it, as long as you are connected to other nodes and can do payments, you are all set.

I guess it will be listed on the explorer in due time, but no idea how often they update their network map.

closed 2018-07-02 by Stadicus