raspibolt/raspibolt #104: Password Authenication

BitcoinVegan, 2018-07-01:
Hello @Stadicus when trying to copy the blockchain for mainnet I ran into an issue running this command
`sudo nano /etc/ssh/sshd_config`

I can't find where to disable the password, I don't know where to do so I have a blank screen, what should go here?
[image]

Stadicus, 2018-07-02:
I think your file `/etc/ssh/sshd_config` is already in use by another process (as noted in the gray line at the bottom). You can either try to kill this process or just do a restart of the Pi.

BitcoinVegan, 2018-07-02:
@Stadicus ok, that worked, but now I don't know where to put "no" for the password authenication.

`#       $OpenBSD: sshd_config,v 1.100 2016/08/15 12:32:04 naddy Exp $

# This is the sshd server system-wide configuration file.  See
# sshd_config(5) for more information.

# This sshd was compiled with PATH=/usr/bin:/bin:/usr/sbin:/sbin

# The strategy used for options in the default sshd_config shipped with
# OpenSSH is to specify options with their default value where
# possible, but leave them commented.  Uncommented options override the
# default value.

#Port 22
#AddressFamily any
#ListenAddress 0.0.0.0
#ListenAddress ::

#HostKey /etc/ssh/ssh_host_rsa_key
#HostKey /etc/ssh/ssh_host_ecdsa_key
#HostKey /etc/ssh/ssh_host_ed25519_key

# Ciphers and keying
#RekeyLimit default none

# Logging
#SyslogFacility AUTH
#LogLevel INFO

# Authentication:

#LoginGraceTime 2m
#PermitRootLogin prohibit-password
#StrictModes yes
#MaxAuthTries 6
#MaxSessions 10

#PubkeyAuthentication yes

`

nipponese, 2018-07-03:
Try adding: `PasswordAuthentication no`

BitcoinVegan, 2018-07-05:
@nipponese @Stadicus where do i add
`PasswordAuthentication`

nipponese, 2018-07-05:
Wherever you want.

Stadicus, 2018-07-07:
Here's my complete `/etc/ssh/sshd_config` for reference:
https://pastebin.com/RTGSb8Dc

nipponese, 2018-07-08:
It looks ok to me, you still can't authenticate with the public key on your local machine?

BitcoinVegan, 2018-07-12:
[image]

I believe I'm confused on whether I add `# PasswordAuthentication no` or if it's suppose to show up in that file and I just disable it. @Stadicus

BitcoinVegan, 2018-07-12:
i found it thanks @Stadicus and @nipponese

closed 2018-07-12 by Stadicus