raspibolt/raspibolt #105: Failing at dphys-swapfile swapon

elkimek, 2018-07-01:
Hi, I'm having similar issues as is in closed #53 

When I run:

```
sudo dphys-swapfile swapon
```

I get:

```
swapon: /mnt/hdd/swapfile: insecure permissions 0775, 0600 suggested.
swapon: /mnt/hdd/swapfile: read swap header failed
```

I tried all abovem mentioned incl.

```
sudo chmod 0600 /mnt/hdd/swapfile
```
and didn't help.

I didn't  have this problem when I was following
[Option 1 (recommended): Format hard disk with Ext4](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#option-1-recommended-format-hard-disk-with-ext4)

But problem appeared first when I went with opt. 2:
[Option 2: Use existing hard disk with NTFS](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#option-2-use-existing-hard-disk-with-ntfs)

Stadicus, 2018-07-02:
This is probably an issue with NTFS as you cannot set very detailed permissions on this file system. Maybe you can get some additional information here:
https://archlinuxarm.org/forum/viewtopic.php?f=18&t=5534

At first, I set up the Pi without swap at all, only adding it at a later stage. It may not be the best idea to have a swap file on the NTFS drive, even if it technically should work... Do you have the option to run the Pi with Ext4?

elkimek, 2018-07-02:
I have, but after some tries where I was solving some issues I found copying synced blockchain faster, because with ext4 it takes 4,5hrs even with SSD.

StephenAsherson, 2018-07-03:
@elkimek I was running into the same 'read swap header failed' issue using HFS+. I "resolved" it by doing the following:

- Increased the swap file size but left the swap file location as default (/var/swap)
- Created the swap file using

`sudo dphys-swapfile setup`

- Copied the swap to my external

`sudo cp /var/swap /mnt/hdd/swapfile`

- Deleted the old swap and updated the dphys-swapfile config to point to /mnt/hdd/swapfile

- Ran the setup again as a sanity (it will check if the swap exists)

`sudo dphys-swapfile setup`

- Enabled the swap

`sudo dphys-swapfile swapon`

I no longer received the error and I can see the swap in the output of the free command

`free -h`

I don't know if there are any side effects to this approach, but I'm going to roll with it for now. Hopefully this helps you out.

sushi30, 2018-07-10:
I used the method given by @StephenAsherson and it also worked. Not aware of any side effects.

masonicboom, 2018-07-12:
@Stadicus this is not exclusive to NTFS. I had this problem with HFS+ as well.

Stadicus, 2018-07-12:
Thanks for the pull request, it's already merged.
See changes here: https://github.com/Stadicus/guides/pull/119

PR #119 (2018-07-12): Fix `swapon: /mnt/hdd/swapfile: read swap header failed`
I was getting this error at the swapon step. Manually creating the swapfile seemed to work. For reference see https://www.raspberrypi.org/forums/viewtopic.php?p=251251.

This fixes #105.
closed 2018-07-15 by Stadicus