raspibolt/raspibolt #106: install bitcoin 0.16.0

SierraLima, 2018-07-04:
this command doesn't install bitcoind

$ sudo install -m 0755 -o root -g root -t /usr/local/bin bitcoin-0.16.0/bin/*

I download and extract all the file but it doesn't work

best regards

TheCHGoon, 2018-07-04:
did you get 16.1?

$ wget https://bitcoin.org/bin/bitcoin-core-0.16.1/bitcoin-0.16.1-arm-linux-gnueabihf.tar.gz

then:

$ sudo install -m 0755 -o root -g root -t /usr/local/bin bitcoin-0.16.1/bin/*

Stadicus, 2018-07-04:
Could be the wrong binary for your platform. Do you use a Raspberry Pi? Did you download the ...arm-linux... file?

SierraLima, 2018-07-05:
Raspberry Pi and downloaded the arm linux file

Stadicus, 2018-07-07:
Might be file system corruption... Here are a few suggetions. Safe bet is to reflash the SD card and start from scratch, however.
https://raspberrypi.stackexchange.com/questions/5404/segmentation-fault

Stadicus, 2018-07-11:
Maybe same issue here? What Raspberry Pi do you use exactly?
https://github.com/Stadicus/guides/issues/115

SierraLima, 2018-07-12:
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

armv6l

CPU revision	: 7

SierraLima, 2018-07-12:
raspberry PI Model B+

Stadicus, 2018-07-12:
So I guess it's a Raspberry Pi 1 B+, right? This would unfortunately mean that it's not powerful enough, and there are no precompiled binaries available for this older (and totally different) cpu platform. See issue https://github.com/Stadicus/guides/issues/115

This guide needs at least a Pi2, I suggest to get the latest P3.

SierraLima, 2018-07-12:
Raspberry pi 3 B+

SierraLima, 2018-07-13:
I reflashed the SD card and started again from scratch but still the same problem

Stadicus, 2018-07-13:
Can you post the output of `cat /proc/cpuinfo`, please?

SierraLima, 2018-07-14:
Raspberry Pi Model B Plus Rev 1.2

processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 900032
Serial		: 0000000080b1ff9b

SierraLima, 2018-07-14:
Thanks so much I interchanged my two raspberry

Stadicus, 2018-07-14:
I had a hunch... 👍

closed 2018-07-14 by Stadicus