raspibolt/raspibolt #108: Can't execute 'apt install linux-image-xu3'

Janaka-Steph, 2018-07-06:
Hello,

I can't execute 'apt install linux-image-xu3'.

root@odroid:~# apt install linux-image-xu3
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-image-xu3

Any ideas?
Thank you.

johneakin, 2018-07-06:
Where is that command even mentioned in the tutorial?

Janaka-Steph, 2018-07-07:
In https://github.com/Stadicus/guides/blob/master/thundroid/README.md#basic-configuration

Ctrl + F can help maybe?

Stadicus, 2018-07-07:
Let's try to be civil. The Thundroid guide is very new and people are trying to help.

I cannot think why this should not work at the moment, but I don't have computer access atm.

Maybe have a look here and double check everything, including that you are using the right Ubuntu image.

https://wiki.odroid.com/odroid-xu4/os_images/linux/ubuntu_4.14/20171212#how_to_update_system_and_kernel_software_don_t_forget_dist-upgrade

Janaka-Steph, 2018-07-07:
Thank you.
Actually I am using Ubuntu 18.04 (ubuntu-18.04-4.14-minimal-odroid-xu4-20180531.img.xz) and if you look at the section "How to Update system and kernel software" here https://wiki.odroid.com/odroid-xu4/os_images/linux/ubuntu_4.14/20180531#how_to_update_system_and_kernel_software_don_t_forget_dist-upgrade  we can notice that `sudo apt install linux-image-xu3` has disappeared.

Stadicus, 2018-07-07:
Good to know. If you deviate from the guide be aware that this has not been tested and might not work at all. Would be interesting to see if you run into other differences, though, maybe add them in this thread for future use?

closed 2018-07-07 by Stadicus