raspibolt/raspibolt #109: Anyone using LNDThinWallet app?

Cees2daBees, 2018-07-07:
Anyone using LNDThinWallet app with their raspibolt. I’m trying to set it up but don’t know what these settings should be or where to find them:

- Host
- TLS cert
- Macaroons hex

Stadicus, 2018-07-07:
Never heard of https://www.lndthinwallet.com/ before, looks interesting. Cannot test, as it's iOS only. Is it Open Source? 

See the Shango bonus guide, it's the same information that's needed to connect.

https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_68_shango.md

Can you share how to connect once you succeeded?

Cees2daBees, 2018-07-15:
I still can't seem to connect to my node. I've attached the settings screen that shows some greyed out text that I guess is there to help the user fill in the information? Does this help explain what I'm supposed to input into these fields?
[image]

Stadicus, 2018-07-15:
Seems like a very complicated way to input your values. I'd say it should be

* Host: `192.168.0.20:10009` (or whatever your ip is)
* TLS cert: run `sudo cat /home/bitcoin/.lnd/tls.cert` and copy paste
* Macaroons hex: run `sudo xxd -ps -u -c 1000 /home/bitcoin/.lnd/admin.macaroon` and copy paste

This should be done using a QR code. Last week I was talking to the guys behind Zap & Shangoo mobile wallets and they agreed on a common format to encode this information. I'll let the developer know.

Stadicus, 2018-07-15:
Sent them the email. As this is not really an issue with this guide, I'll close this issue and ask you to talk to the guys behind LNDThinWallet directly.

closed 2018-07-15 by Stadicus